voed/esphome_prana_ble #1: Entities not shown in HA

okruts, 2024-08-04:
I followed your example, new device was discovered but no entities are shown in Home Assistant except firmware. Previously I tried hassio-prana integration and was able to connect using Bluetooth Proxy. Can you please guide me what I'm doing wrong? . Here is my yaml:

```
esphome:
  name: prana
  friendly_name: prana

# Strongly recommended to use IDF framework instead of arduino.
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "MY_ENCRYPTION_KEY"

ota:
  - platform: esphome
    password: "OTA_PASSWORD"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Prana Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# MAC address can be obtained via "BLE Scanner app on your smartphone or BLE Client Tracker component"
# TODO: try to merge it with "prana_ble"
ble_client:
  - mac_address: MY_PRANA_BLE_MAC_ADDRESS
    id: prana

prana_ble:
  id: prana_client
  ble_client_id: prana

# voltage and frequency sensors should be accessible for all models
sensor:
  - platform: prana_ble
    prana_ble_id: prana_client
    voltage:
      name: "Prana voltage"
    frequency:
      name: "Prana frequency"

switch:
# Switch to turn off BLE connection. Useful if you want to connect from native smartphone app
  - platform: prana_ble
    prana_ble_id: prana_client
    connect:
      name: prana_connect
      restore_mode: ALWAYS_ON
    enable:
      id: enable
      name: prana_enable
    heating:
      id: heating
      name: prana_heating
    winter_mode:
      name: prana_winter_mode

# TODO: make option to control in/out fans
fan:
  - platform: prana_ble
    prana_ble_id: prana_client
    name: prana_fan

external_components:
  - source: github://voed/esphome_prana_ble
    components: [prana_ble]
```

voed, 2024-08-04:
Hi. Check the logs of your device from the ESPHome addon. You should be able to see if it receiving any data from your recuperator:
[image]

Also i am not sure if it works on esp32-c3 board, i will check it later.

voed, 2024-08-05:
Well, for some reason idf framework does not work at all on C3 board. So try with arduino framework, it works for me but i am not sure if it is stable.

okruts, 2024-08-06:
I managed to run it with arduino framework. But it works very poorly. All sensor data is wrong or not shown at all (I'm using prana 150 Eco life series). I will try to use classic esp32 as you suggested later on instead of c3. 

Btw is it possible to add switch for auto and auto+ modes? That ones I use mostly with my recuperator.

[image]

voed, 2024-08-06:
That's good news. I have a very old model without any sensors so i was not able to test that. I have added some logs so please change log level to `VERBOSE` and provide this logs:
[image]

Same for auto and auto+, turn on that mode and provide the packet string with it because i don't have such modes on my recuperator.

Also you need to add `refresh: 1s` to the external_components to make sure it will install the new version from this repo.

okruts, 2024-08-06:
Here you go. 

With auto mode:

```
[17:55:33][V][prana_ble:404]: Packet: BE.EF.05.01.66.B2.8D.C4.5A.C0.01.C0.20.C0.00.C0.01.C0.00.00.01.C0.00.40.01.40.0A.C0.01.C0.0A.C0.01.C0.0A.00.00.80.00.00.00.C0.00.00.01.00.00.00.81.14.00.80.D6.00.81.09.00.80.E4.00.B2.81.90.80.79.00.E3.00.00.00.00.00.00.00.00.00.00.82.DC.00.00.00.00.00.00.00.80.00.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.E9.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
[17:55:33][V][prana_ble:405]: speed: 1
[17:55:33][V][prana_ble:406]: speed: 1
[17:55:33][V][prana_ble:407]: speedIn: 1
[17:55:33][D][prana_ble:116]: [prana_enable] update()
[17:55:33][D][prana_ble:116]: [prana_heating] update()
[17:55:34][D][prana_ble:092]: [prana_fan] update()
[17:55:34][D][prana_ble:226]: BEEF0501000000005A
[17:55:34][V][prana_ble:049]: [prana_fan] Handling on_status with data=0x3fca5ab4
[17:55:34][D][prana_ble:096]: [prana_fan] update_status result=true
[17:55:34][D][prana_ble:116]: [prana_winter_mode] update()
[17:55:34][D][prana_ble:116]: [prana_connect] update()
[17:55:34][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[17:55:34][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=4 gattc_if=3
[17:55:34][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_WRITE_CHAR_EVT
[17:55:34][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:55:34][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=10 gattc_if=3
[17:55:34][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_NOTIFY_EVT
[17:55:34][D][prana_ble:398]: [PRANA_MAC] Notifying 6 children of latest status @0x3fcc37bc.
[17:55:34][V][prana_ble:024]:  Handling on_status with data=0x3fcc37bc
[17:55:34][V][sensor:043]: 'inside temperature': Received new state 20.000000
[17:55:34][D][sensor:094]: 'inside temperature': Sending state 20.00000 °C with 1 decimals of accuracy
[17:55:34][V][sensor:043]: 'outside temperature': Received new state 10.000000
[17:55:34][D][sensor:094]: 'outside temperature': Sending state 10.00000 °C with 1 decimals of accuracy
[17:55:34][V][sensor:043]: 'pressure': Received new state 220.000000
[17:55:34][D][sensor:094]: 'pressure': Sending state 220.00000 hPa with 1 decimals of accuracy
[17:55:34][V][sensor:043]: 'tvoc': Received new state 29568.000000
[17:55:34][D][sensor:094]: 'tvoc': Sending state 29568.00000 ppb with 0 decimals of accuracy
[17:55:34][V][sensor:043]: 'co2': Received new state -28543.000000
[17:55:34][D][sensor:094]: 'co2': Sending state -28543.00000 ppm with 0 decimals of accuracy
[17:55:34][V][sensor:043]: 'voltage': Received new state 0.000000
[17:55:34][D][sensor:094]: 'voltage': Sending state 0.00000 V with 0 decimals of accuracy
```
 
With Auto+ mode:

```
[18:06:12][V][prana_ble:404]: Packet: BE.EF.05.01.66.B2.90.42.5A.C0.01.C0.20.C0.00.C0.01.C0.00.00.02.C0.00.40.01.40.0A.C0.01.C0.0A.C0.01.C0.0A.00.00.80.00.00.00.C0.00.00.01.00.00.00.81.15.00.80.D6.00.81.0B.00.80.E4.00.B2.81.90.80.41.00.E3.00.00.00.00.00.00.00.00.00.00.82.DC.00.00.00.00.00.00.00.80.00.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.E9.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
[18:06:12][V][prana_ble:405]: speed: 1
[18:06:12][V][prana_ble:406]: speed: 1
[18:06:12][V][prana_ble:407]: speedIn: 1
[18:06:12][V][prana_ble:408]: speedIn: 1
[18:06:12][D][prana_ble:116]: [prana_enable] update()
[18:06:13][D][prana_ble:116]: [prana_heating] update()
[18:06:13][D][prana_ble:092]: [prana_fan] update()
[18:06:13][D][prana_ble:244]: Last update was 31 ms ago, skipping update request
[18:06:13][V][prana_ble:049]: [prana_fan] Handling on_status with data=0x3fca5ab4
[18:06:13][D][prana_ble:096]: [prana_fan] update_status result=true
[18:06:13][D][prana_ble:116]: [prana_winter_mode] update()
[18:06:13][D][prana_ble:116]: [prana_connect] update()
[18:06:13][D][prana_ble:116]: [prana_enable] update()
[18:06:14][D][prana_ble:116]: [prana_heating] update()
[18:06:14][D][prana_ble:092]: [prana_fan] update()
[18:06:14][D][prana_ble:244]: Last update was 6 ms ago, skipping update request
[18:06:14][V][prana_ble:049]: [prana_fan] Handling on_status with data=0x3fca5ab4
[18:06:14][D][prana_ble:096]: [prana_fan] update_status result=true
[18:06:14][D][prana_ble:116]: [prana_winter_mode] update()
[18:06:14][D][prana_ble:116]: [prana_connect] update()
[18:06:14][D][prana_ble:116]: [prana_enable] update()
[18:06:14][D][prana_ble:116]: [prana_heating] update()
[18:06:14][V][esp32_ble:323]: (BLE) gap_event_handler - 3
[18:06:14][V][esp32_ble:323]: (BLE) gap_event_handler - 3
[18:06:15][V][esp32_ble:323]: (BLE) gap_event_handler - 3
[18:06:15][D][prana_ble:092]: [prana_fan] update()
[18:06:15][D][prana_ble:226]: BEEF0501000000005A
[18:06:15][V][prana_ble:049]: [prana_fan] Handling on_status with data=0x3fca5ab4
[18:06:15][D][prana_ble:096]: [prana_fan] update_status result=true
[18:06:15][D][prana_ble:116]: [prana_winter_mode] update()
[18:06:15][D][prana_ble:116]: [prana_connect] update()
[18:06:15][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[18:06:15][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=4 gattc_if=3
[18:06:15][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_WRITE_CHAR_EVT
[18:06:15][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[18:06:15][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=10 gattc_if=3
[... cut ...]
[18:06:15][V][sensor:043]: 'outside temperature': Received new state 11.000000
[18:06:15][D][sensor:094]: 'outside temperature': Sending state 11.00000 °C with 1 decimals of accuracy
[18:06:15][V][sensor:043]: 'pressure': Received new state 220.000000
[18:06:15][D][sensor:094]: 'pressure': Sending state 220.00000 hPa with 1 decimals of accuracy
[18:06:15][V][sensor:043]: 'tvoc': Received new state 16256.000000
```

I also noticed that packets are a bit different each time I switch mode back and force between auto and auto+.

voed, 2024-08-06:
Thanks, i will check it.

> I also noticed that packets are a bit different each time I switch mode back and force between auto and auto+.

It's okay, the packet contains full state of the device, including temperatures and other sensors, so they should be different.

voed, 2024-08-06:
Sorry, can you do that again, but also attach the screenshots from the Prana app? I want to compare sensors values. You can PM me in telegram `@voed1337` if you wish.

okruts, 2024-08-07:
Auto mode:

```
[08:49:36][V][prana_ble:404]: Packet: BE.EF.05.01.66.B3.5F.4E.5A.C0.01.C0.20.C0.00.C0.01.C0.00.00.01.C0.00.40.01.40.0A.C0.01.C0.0A.C0.01.C0.0A.00.00.80.00.00.00.C0.00.00.01.00.00.00.81.33.00.81.38.00.81.29.00.81.37.00.AC.81.BC.80.D1.00.00.00.00.00.00.00.00.00.00.00.00.82.DD.00.00.00.00.00.00.00.80.00.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.E9.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
[08:49:36][V][esp32_ble:323]: (BLE) gap_event_handler - 3

.......

[08:49:42][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[08:49:42][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=4 gattc_if=3
[08:49:42][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_WRITE_CHAR_EVT
[08:49:42][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[08:49:42][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=10 gattc_if=3
[08:49:42][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_NOTIFY_EVT
[08:49:42][D][prana_ble:398]: [PRANA_MAC] Notifying 6 children of latest status @0x3fcc3d48.
[08:49:42][V][prana_ble:024]:  Handling on_status with data=0x3fcc3d48
[08:49:42][V][sensor:043]: 'outside temperature': Received new state 41.000000
[08:49:42][D][sensor:094]: 'outside temperature': Sending state 41.00000 °C with 1 decimals of accuracy
[08:49:42][V][sensor:043]: 'pressure': Received new state 221.000000
[08:49:42][D][sensor:094]: 'pressure': Sending state 221.00000 hPa with 1 decimals of accuracy
[08:49:42][V][sensor:043]: 'tvoc': Received new state -10880.000000
[08:49:42][D][sensor:094]: 'tvoc': Sending state -10880.00000 ppb with 0 decimals of accuracy
[08:49:42][V][sensor:043]: 'co2': Received new state -16767.000000
[08:49:42][D][sensor:094]: 'co2': Sending state -16767.00000 ppm with 0 decimals of accuracy
[08:49:43][V][sensor:043]: 'voltage': Received new state 0.000000
```
[image]

Auto+ mode:

```
[08:54:59][V][prana_ble:404]: Packet: BE.EF.05.01.66.B3.60.90.5A.C0.01.C0.20.C0.00.C0.01.C0.00.00.02.C0.00.40.01.40.0A.C0.01.C0.0A.C0.01.C0.0A.00.00.80.00.00.00.C0.00.00.01.00.00.00.81.35.00.81.35.00.81.2A.00.81.37.00.AB.81.BD.80.E6.00.00.00.00.00.00.00.00.00.00.00.00.82.DD.00.00.00.00.00.00.00.80.00.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.E9.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
[08:54:59][V][prana_ble:405]: speed: 1
[08:54:59][V][prana_ble:406]: speed: 1
[08:54:59][V][prana_ble:407]: speedIn: 1
[08:54:59][V][prana_ble:408]: speedIn: 1
[08:54:59][V][prana_ble:409]: speedOut: 1
[08:54:59][V][prana_ble:410]: speedOut: 1
[08:54:59][V][prana_ble:411]: heat: 0
[08:54:59][V][prana_ble:412]: heat: 0
[08:54:59][V][prana_ble:413]: winter_mode: 0
[08:54:59][V][prana_ble:414]: winter_mode: 0
[08:54:59][V][prana_ble:415]: winter_mode: 0
[08:54:59][V][prana_ble:416]: voltage: 0
[08:54:59][W][component:237]: Component esp32_ble took a long time for an operation (57 ms).
[08:54:59][W][component:238]: Components should block for at most 30 ms.

.......

[08:55:04][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[08:55:04][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=4 gattc_if=3
[08:55:04][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_WRITE_CHAR_EVT
[08:55:04][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[08:55:04][V][esp32_ble_client:121]: [0] [PRANA_MAC] gattc_event_handler: event=10 gattc_if=3
[08:55:04][D][esp32_ble_client:110]: [0] [PRANA_MAC] ESP_GATTC_NOTIFY_EVT
[08:55:04][D][prana_ble:398]: [PRANA_MAC] Notifying 6 children of latest status @0x3fcc42dc.
[08:55:04][V][prana_ble:024]:  Handling on_status with data=0x3fcc42dc
[08:55:04][V][sensor:043]: 'outside temperature': Received new state 42.000000
[08:55:04][D][sensor:094]: 'outside temperature': Sending state 42.00000 °C with 1 decimals of accuracy
[08:55:04][V][sensor:043]: 'pressure': Received new state 221.000000

.......

[08:55:09][V][sensor:043]: 'outside temperature': Received new state 42.000000
[08:55:09][D][sensor:094]: 'outside temperature': Sending state 42.00000 °C with 1 decimals of accuracy
[08:55:09][V][sensor:043]: 'pressure': Received new state 221.000000
[08:55:09][D][sensor:094]: 'pressure': Sending state 221.00000 hPa with 1 decimals of accuracy
[08:55:09][V][sensor:043]: 'tvoc': Received new state -6784.000000
[08:55:09][D][sensor:094]: 'tvoc': Sending state -6784.00000 ppb with 0 decimals of accuracy
[08:55:09][V][sensor:043]: 'co2': Received new state -16767.000000
[08:55:09][D][sensor:094]: 'co2': Sending state -16767.00000 ppm with 0 decimals of accuracy
[08:55:09][V][sensor:043]: 'voltage': Received new state 0.000000
[08:55:09][D][sensor:094]: 'voltage': Sending state 0.00000 V with 0 decimals of accuracy
[08:55:09][V][sensor:043]: 'frequency': Received new state 0.000000
[08:55:09][D][sensor:094]: 'frequency': Sending state 0.00000 Hz with 0 decimals of accuracy
```

[image]

Screenshots from app taken in the moment when packet was received

voed, 2024-08-07:
Thanks! 
Here's what i got for now:
1. Your device does not provide voltage & frequency data, so you can just remove that sensors.
2. Pressure and humidity should be fixed now. Maybe VOC and CO2 too, needs to be tested.
3. Auto(not auto+) mode switch should work. I see some duplicates of switches on your first screenshot, so try to remove the device from HA and add it again. If it still does not work please show your current configuration. For auto+ please create another issue, it requires some more actions from you to be implemented.
4. Temperature sensors still needs some work, but should be possible to fix

okruts, 2024-08-08:
Great progress! VOC, CO2 and pressure now show correct values. Humidity still in unknown status. Also Auto Mode switch did not appear in HA. I double checked that I added auto mode switch my template, recompile it few times, re-added device in HA, still nothing
[image]
[image]

Also heating and winter mode doesn't work for me. After I turn any of those switches it immediately back to turn off state in the same time my recuperator turn off current mode and in a few seconds it switches to the night mode (I guess because it is first in the list). It even caused a bug in recuperator that I was unable to select any mode, it automatically switch back to Night Mode until I completely shutdown Prana

Btw this is a list of modes that I have:
[image]

voed, 2024-08-08:
That's really weird. Can you check if fan control works for you? I think your model has different commands for control, if so, it will be really difficult to make it work:(

okruts, 2024-08-08:
Yes, fan and prana_enable works. I know that my model is using esp32 inside, so it's possible to control it via bluetooth or wifi.

voed, 2024-08-08:
I think the only way to figure it out is to provide Bluetooth HCI logs. You can check here how to do that:
https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa
Enable logging, connect to the recuperator via bluetooth, press that winter/heating/auto/auto+ buttons, disable logger and get the log file. Also provide at least half of your recuperator's mac address, i need it to filter out the logs.

voed, 2024-08-08:
Found a problem with auto mode switch, now it should be available, but not sure if it will work for you

okruts, 2024-08-08:
It works!! And humidity now works as well! Thank you 

I will create separate issue for auto+ mode.

Do you have display on your recuperator? There is possibility to show different sensor information on mine, it's configurable either via app or remote. Do you think it's possible to control this with your component? It will be cool for automation to display some default information but in case of abnormal CO2 or VOC it will show those values instead.

[image]

voed, 2024-08-08:
No, i have like 2016 year model with 7-segment display only, it does not even work with new Prana Online app. I don't think this is possible on stock firmware, can't say more without owning that type of recuperator:)

closed 2024-08-08 by voed