voed/esphome_prana_ble #2: Ability to connect 1+ devices

Trteich, 2024-08-06:
Hi, thank you very much for your work, it is exactly what I was searching for.
Is there ability to connect more than 1 prana recuperator? If yes, how config file should look like?

Trteich, 2024-08-07:
I've already dealed with it. Works perfect. As mentioned on ESPHome web site, esp32 can serve up to 3 devices without issues

voed, 2024-08-07:
Ok, try it:
``` yaml

ble_client:
  - mac_address: 00:A0:50:00:00:03
    id: prana
  - mac_address: FF:FF:FF:FF:FF:FF
    id: prana2

prana_ble:
  - id: prana_client
    ble_client_id: prana

  - id: prana_client2
    ble_client_id: prana2

sensor:
  - platform: prana_ble
    prana_ble_id: prana_client
    voltage:
      name: "Prana voltage"
    frequency:
      name: "Prana frequency"

  - platform: prana_ble
    prana_ble_id: prana_client2
    voltage:
      name: "Prana2 voltage"
    frequency:
      name: "Prana2 frequency"
```

Trteich, 2024-08-07:
here is my config which works with 2 devices

```shell
esphome:
  name: prana
  friendly_name: prana

# Strongly recommended to use IDF framework instead of arduino.
esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "CODE"

ota:
  - platform: esphome
    password: "PASSWD"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Prana Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# MAC address can be obtained via "BLE Scanner app on your smartphone or BLE Client Tracker component"
# TODO: try to merge it with "prana_ble"
ble_client:
  - mac_address: MAC1
    id: PranaFirstFloor
  - mac_address: MAC2
    id: PranaSecondFloor

prana_ble:
  - id: prana_first_floor
    ble_client_id: PranaFirstFloor
  - id: prana_second_floor
    ble_client_id: PranaSecondFloor

# voltage and frequency sensors should be accessible for all models
sensor:
  - platform: prana_ble
    prana_ble_id: prana_first_floor
    voltage:
      name: "Prana1 voltage"
    frequency:
      name: "Prana1 frequency"
  - platform: prana_ble
    prana_ble_id: prana_second_floor
    voltage:
      name: "Prana second floor voltage"
    frequency:
      name: "Prana prana second floor frequency"

switch:
# Switch to turn off BLE connection. Useful if you want to connect from native smartphone app
  - platform: prana_ble
    prana_ble_id: prana_first_floor
    connect:
      name: connect_first_floor
      restore_mode: ALWAYS_ON
    enable:
      id: enable_first_floor
      name: enable_first_floor
    heating:
      id: heating_first_floor
      name: heating_first_floor
    winter_mode:
      name: winter_mode_first_floor
  - platform: prana_ble
    prana_ble_id: prana_second_floor
    connect:
      name: connect_second_floor
      restore_mode: ALWAYS_ON
    enable:
      id: enable_second_floor
      name: enable_second_floor
    heating:
      id: heating_second_floor
      name: heating_second_floor
    winter_mode:
      name: winter_mode_second_floor

# TODO: make option to control in/out fans
fan:
  - platform: prana_ble
    prana_ble_id: prana_first_floor
    name: first_floor
  - platform: prana_ble
    prana_ble_id: prana_second_floor
    name: second_floor

external_components:
  - source: github://voed/prana_ble@main
    components: [prana_ble]
    refresh: 30s
```

voed, 2024-08-08:
Thanks for feedback.

closed 2024-08-08 by voed