voed/esphome_prana_ble #3: Tvoc sensor can't be added

okruts, 2024-08-06:
When I'm trying to build esphome with tvoc sensor icluded I received this error:

```
Compiling .pioenvs/prana/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:353:33: error: 'class esphome::prana_ble::PranaBLESensors' has no member named 'set_voc'; did you mean 'set_tvoc'?
   prana_ble_pranablesensors_id->set_voc(sensor_sensor_id_8);
                                 ^~~~~~~
                                 set_tvoc
*** [.pioenvs/prana/src/main.cpp.o] Error 1
```

YAML:

```
...

sensor:
  - platform: prana_ble
    prana_ble_id: prana_client
    tvoc:
      name: "Prana tvoc"

...
```

voed, 2024-08-06:
Should be fixed now. Thanks.

okruts, 2024-08-06:
Yes, was able to add. Thank you

closed 2024-08-06 by okruts