voed/esphome_prana_ble #5: Add Auto+ mode switch

okruts, 2024-08-08:
In the latest Prana recuperators they added Auto and Auto+ modes which automatically adjust fan speed depending on sensor information

Auto+ mode limits fan speed to 3 so it more suitable for night time.

voed, 2024-08-09:
Ok, so the recuperator receives all feedback by a simple "commands". Without knowing this commands i am not able to implement this feature.
There are two ways to obtain this command. The first one is to get HCI logs as i mentioned in https://github.com/voed/esphome_prana_ble/issues/1#issuecomment-2276014104

The second way is to "bruteforce" the recuperator and find needed command.
Open "BLE Scanner" app for android, connect to your recuperator, click on "custom service", click on "W"(write), select "Byte array". Enter `beef0401` this is the command to turn off the recuperator, just to make sure you made everything right. If it worked, turn it on and try this commands:
`beef0403` 
`beef0404` 
`beef0408` 
`beef0413` 
`beef0414` 
`beef0415` 
`beef0417`
`beef0419` 
`beef041a` 
`beef041b` 
`beef041c` 
I don't think they can damage your device somehow, but you know, for your own risk as there may be some service commands or something. Also getting HCI logs is preferred way because new commands may be completely different from this.

okruts, 2024-08-11:
Okay so I did some "bruteforce" as you suggested. I tested rang of commands from `beef0401` to `beef041f` and discovered most of the functionality of my recuperator. Most of the commands basically mimics prana remote and act a bit differently from prana app. So for the auto+ mode I didn't found separate command as we have in the app but there is toggle command that switch between auto/auto+/auto off

Here is the list of all commands that I discovered:
REMOTE - command act the same as on remote
APP - command act the same as on the app

`beef0402` - brightness toggle 0-6 --- REMOTE
`beef0406` - night mode --- REMOTE
`beef0407` - boost mode --- REMOTE
`beef0409` - control of air flows (lock/unlock) --- APP (probably useless)
`beef040a` - turn on/turn off both fans --- REMOTE
`beef040b` - all fans - --- APP
`beef040c` - all fans + --- APP
`beef040d` - turn off/on air supply --- REMOTE 
`beef040e` -  air supply + --- APP
`beef040f` - air supply - --- APP
`beef0410` - turn off/on air extraction --- REMOTE 
`beef0411` - air extraction + --- APP
`beef0412` - air extraction - --- APP
`beef0413` - sleep timer on/off --- APP
`beef0414` - decrese sleep timer (-10 minutes) --- REMOTE
`beef0415` - increase sleep timer (+10 minutes) --- REMOTE

`beef0418` - auto toogle -- auto/auto+/auto off --- REMOTE

`beef0419` - display left --- REMOTE
`beef041a` - display right --- REMOTE

Display indicators from left to right DATE - TIME - FAN - CO2 - TVOC - ATM - HUMIDITY - TEMP OUT - TEPM IN
Maybe somehow it's possible to select specific display indicator if we have only `display left / display right` commands

I guess there is probably more commands to specifically switch between auto and auto+ mode or it's possible to do some workaround to select specific mode with this toggle command.

Remote:
[image]

okruts, 2024-08-11:
I also tried commands from `beef0420` there is a bunch of commands to specifically set fan speed from 0 to 10 for each specific air flow - Extraction, Supply or Both

voed, 2024-08-11:
Oh, i see. It would not be hard to implement auto+ mode then, but i think i will move it to the fan control instead of switches.

> Maybe somehow it's possible to select specific display indicator if we have only display left / display right commands

Try to cycle between them and provide the packet log strings for each mode. If packet contains any info about what is currently displayed, it should be possible to do that. But create a separate issue for that please:)

> I also tried commands from beef0420 there is a bunch of commands to specifically set fan speed from 0 to 10 for each specific air flow - Extraction, Supply or Both

Interesting discovery, might be useful. Thanks!

okruts, 2024-08-12:
I tested carefully today current implementation of auto mode switch. Basically it works now as I described, each time I toggle switch it change auto mode from auto to auto+ to auto off doesn't matter in which way I toggle switch on or off

voed, 2024-08-15:
I have created a separate branch, please test it:
```

okruts, 2024-08-15:
I see new preset button appear but can't test it as it almost immediately crashing after boot

```
WARNING Can't connect to ESPHome API for prana @ 10.10.0.70: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.10.0.70', port=6053))]: [Errno 111] Connect call failed ('10.10.0.70', 6053) (SocketAPIError)
INFO Trying to connect to prana @ 10.10.0.70 in the background

01:15:07][D][esp32_ble_client:110]: [0] [
] ESP_GATTC_REG_FOR_NOTIFY_EVT
[01:15:07][W][esp32_ble_client:275]: [0] [] esp_ble_gattc_get_descr_by_char_handle error, status=10
[01:15:07][D][prana_ble:490]: [] wrote notify=true to status config 0x002b, for conn 0
[01:15:07][W][component:237]: Component esp32_ble took a long time for an operation (72 ms).
[01:15:07][W][component:238]: Components should block for at most 30 ms.
[01:15:07][D][prana_ble:116]: [power] update()
[01:15:07][V][esp32_ble:351]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[01:15:07][V][esp32_ble_client:121]: [0] [] gattc_event_handler: event=10 gattc_if=3
[01:15:07][D][esp32_ble_client:110]: [0] [] ESP_GATTC_NOTIFY_EVT
[01:15:07][D][prana_ble:416]: [] Notifying 5 children of latest status @0x3fcb0130.
[01:15:07]

```

voed, 2024-08-15:
Is that on IDF framework?

okruts, 2024-08-18:
Yes. It seems it started working, just need longer time to boot. Auto, Auto+ buttons works but not manual. Manual can either turn on auto, auto+ or turn off, as button on a remote

voed, 2024-08-23:
Try to check now, i have added some logs. If problem still exists, remove all switches and sensors from the config, leave only fan entity, then click on modes few times and provide logs.

matlar83, 2024-08-26:
I tested it too.
I can see the presets and changing preset works perfectly for me.

Instead, when the auto or auto+ mode is set, either from HA or with the remote, the fan speed is not aligned with the device anymore.
Steps to reproduce:
- set preset = manual from HA
- change speed to 4/5 with the remote --> speed is updated in HA
- set preset = auto or auto+ from HA or from remote --> speed changes to 1/5 on the device, but on HA fan speed is still 4/5

voed, 2024-08-27:
Can you check this behavior in the Prana app? If it shows the correct fan speed during auto/auto+ mode please share your log with data packets for manual/auto/auto+ modes.

matlar83, 2024-08-27:
Sure, I'll check it when I get home.
How can I log data?

voed, 2024-08-27:
From esphome addon. See here for details: 
https://github.com/voed/esphome_prana_ble/issues/1#issuecomment-2267809253

matlar83, 2024-08-29:
Here the logs: 
[logs_test-dev_logs.txt](https://github.com/user-attachments/files/16795175/logs_test-dev_logs.txt)

Theese are the operations I did:
1) set preset = manual from HA
2) change speed to 5/5 from HA
3) set preset = auto with the remote --> speed changed to 1/5 on the device, but on HA fan speed is still 5/5

Let me know if you need anything else.
Thank you

voed, 2024-08-29:
Sorry, i forgot, you need to recompile your firmware with verbose log level
```
logger:
  level: VERBOSE
```
Then you'll see the packet string in the log:
[image]

But for now i think it may be related to #7 because auto mode changes speeds for the supply/extraction fans separately

matlar83, 2024-08-29:
No problem, here it is the verbose logs: 
[logs_test-dev_logs.txt](https://github.com/user-attachments/files/16800149/logs_test-dev_logs.txt)

voed, 2024-08-31:
As i thought, the auto mode unlocks the fans and changes their speed separately, so this should be fixed when i deal with #7

voed, 2024-09-04:
@okruts still waiting for a feedback from you.

closed 2024-11-05 by voed