voed/esphome_prana_ble #7: Add switches for Air Supply and Air Extraction

okruts, 2024-08-12:
As per previous discovery we have specific commands to disable Air Supply / Air Extraction.

It can be useful to turn off Air Supply for example when Air Quality outside is bad. I don't know why prana is not implemented this as my recuperator is often goes in a loop (in auto mode) when air outside is bad and recuperator is intaking air and immediately trying to extract it on max fan speed as it has only one inside VOC sensor 

ble commands:
`beef040d` - turn off/on air supply 
`beef0410` - turn off/on air extraction

voed, 2024-08-12:
It is planned to implement separate controls for intake/extract fans, but for now i don't know how to make that without confusing what is actually controlled. It may be done only by adding two more fan entities, so there will be one fan to control both fans, one fan to control intake fan, one fan to control extract fan and a switch entity to switch between separate controlling. And looks like there are no way to mark fan unavailable so all 3 of them will be active all the time.
But if it's really necessary for you, i can implement the switches for now.

matlar83, 2024-08-23:
I want, first of all, to say thank you for this amazing project! I'm currently testing it with success with Prana ERP 160 and Prana ERP 210.

For this specific topic,
I suggest to have just two fans (intake/extract) + 1 switch to lock the two fans speed.
If the switch is off, the two fans are independent, if it is on instead, the two fans are locked and changes to one fan speed result in the same speed applied also to the other one.
Does it make sense to you?

voed, 2024-08-23:
I will think about it, thanks

PR #13 (2024-10-21): Implement separate fan control
Now fans may be controlled separately by using fan_lock switch.

``` yaml
switch:
  - platform: prana_ble
    prana_ble_id: prana_client
    connect:
       name: prana_connect
       restore_mode: ALWAYS_ON
    enable:
       id: enable
       name: prana_enable
    heating:
      id: heating
      name: prana_heating
    winter_mode:
      name: prana_winter_mode
    fan_lock:
      name: prana_fan_lock
      

fan:
  - platform: prana_ble
    prana_ble_id: prana_client
    has_auto: true
    name: both_fans
    fan_type: BOTH

  - platform: prana_ble
    prana [... cut]
closed 2024-10-21 by voed