voed/esphome_prana_ble #8: Add display backlight control

matlar83, 2024-08-26:
It would be useful to add a way to control the device display backlight.
On the app it is possible to set it with a slider: values in the range 1..6
[image]

I would be more than happy to help testing this new feature

vpritulin, 2024-10-18:
@voed Hello! 
I managed to achieve that it was possible to control brightness. Maybe the code is not perfect, but it already works. But so far I have not managed to achieve the result to get the current state of brightness.
You can look at the code example

https://github.com/vpritulin/esphome_prana_ble/commit/f36663d26a2faae6d6702c5c4e7d84a3a09ff79b
https://github.com/vpritulin/esphome_prana_ble/commit/5e529e59a34412db702ef10faa9aa16605dd5f26

and in yaml file 

```
number:
  - platform: template
    name: "Brightness Level"
    id: brightness_level
    min_value: 1
    max_value: 6
    step: 1
    set_action:
      - lambda: |-
          auto prana_hub = id(prana_client);
          uint8_t brightness = (uint8_t) id(brightness_level).state;
          
          uint8_t mapped_brightness;
          switch (brightness) {
            case 0:
              mapped_brightness = 0;
              break;
            case 1:
              mapped_brightness = 1;
              break;
            case 2:
              mapped_brightness = 2;
              break;
            case 3:
              mapped_brightness = 3;
              break;
            case 4:
              mapped_brightness = 4;
              break;
            case 5:
              mapped_brightness = 5;
              break;
            case 6:
              mapped_brightness = 6; 
              break;
          }
          
          if (prana_hub->set_brightness(mapped_brightness)) {
            ESP_LOGI("BLE", "Brightness set to %d (mapped to %d)", brightness, mapped_brightness);
          } else {
            ESP_LOGE("BLE", "Failed to set brightness");
```

voed, 2024-10-19:
Your method will not work, at least for my old recuperator, because brightness command is cycling through brightness levels and does not support directly setting it.

vpritulin, 2024-10-19:
Got you. I tested it on my Prana 200C Eco Life M2023 and it worked.

voed, 2024-10-21:
Please check now, should work fine

vpritulin, 2024-10-21:
@voed The brightness itself works, but with the latest update the humidity is not displayed correctly and there is a problem with some temperature sensor. The connection is no longer stable, it is connected and then disconnected
[image]

voed, 2024-10-21:
Please create a new issue for that

closed 2024-10-21 by voed