voed/esphome_prana_ble #12: Outside Temperature

vpritulin, 2024-10-17:
Prana 200c eco life no information about outside temperature at the outlet. In homeassistant it is displayed 0
[image]

vpritulin, 2024-10-17:
Also on the screenshot you can see that the voltage and frequency are 0, but this isn't so important for me

voed, 2024-10-17:
Hi. Voltage and frequency are only available for some old models so you can just remove them from the config.
For a temperature please change log level to verbose and provide packet log(as described [here](https://github.com/voed/esphome_prana_ble/issues/5#issuecomment-2317623861))

vpritulin, 2024-10-17:
@voed My log file
[logs_prana-rec_logs.txt](https://github.com/user-attachments/files/17421918/logs_prana-rec_logs.txt)

vpritulin, 2024-10-17:
I look in the application, there are only 3 temperatures, so I think this is normal for this model. Maybe on other versions the recuperator has 4 of them
[image]

voed, 2024-10-18:
I see. Yes, according to logs your device has only one outside sensor. Just remove it from the config then.

closed 2024-10-18 by voed