voed/esphome_prana_ble #14: unstable connection after update and wrong data

vpritulin, 2024-10-21:
The brightness itself works, but with the latest update the humidity is not displayed correctly and there is a problem with some temperature sensor. The connection is no longer stable, it is connected and then disconnected
[image]

voed, 2024-10-21:
It looks like some memory corruption maybe esp is out of memory.  If you have any other components in your config try to remove them and if you have more than one devices, leave just one. If that does not help, change the log level to verbose, provide the logs and your configuration.

vpritulin, 2024-10-21:
I have esp 32 wroom, no sensors. When installing it says that the memory is full on 87.7, but I think it should be enough.
RAM:   [==        ]  16.5% (used 53940 bytes from 327680 bytes)
Flash: [========= ]  87.7% (used 1608393 bytes from 1835008 bytes)

After install and I can't get animemore logs
```
[17:31:49][C][wifi:451]:   DNS2: 192.168.0.1
[17:31:49][D][prana_ble:175]: [Inlet fan] update()
[17:31:50][C][logger:185]: Logger:
[17:31:50][C][logger:186]:   Level: VERBOSE
[17:31:50][C][logger:188]:   Log Baud Rate: 115200
[17:31:50][C][logger:189]:   Hardware UART: UART0
[17:31:50][C][esp32_ble:391]: ESP32 BLE:
[17:31:50][C][esp32_ble:393]:   MAC address: 00:00:00:00:00:00
[17:31:50][C][esp32_ble:394]:   IO Capability: none
[17:31:50][D][prana_ble:118]: [Fan lock] update()
[17:31:50][C][ble_client:027]: BLE Client:
[17:31:50][C][ble_client:028]:   Address: 00:00:00:00:00:00
[17:31:50][C][ble_client:029]:   Auto-Connect: TRUE
[17:31:50][D][prana_ble:118]: [Heating] update()
[17:31:50][C][esp32_ble_tracker:655]: BLE Tracker:
[17:31:50][C][esp32_ble_tracker:656]:   Scan Duration: 300 s
[17:31:50][C][esp32_ble_tracker:657]:   Scan Interval: 320.0 ms
[17:31:50][C][esp32_ble_tracker:658]:   Scan Window: 30.0 ms
WARNING prana @ 192.168.0.165: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for prana @ 192.168.0.165
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for prana @ 192.168.0.165: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.0.165', port=6053))]: [Errno 111] Connect call failed ('192.168.0.165', 6053) (SocketAPIError)
INFO Trying to connect to prana @ 192.168.0.165 in the background
INFO Successfully connected to prana @ 192.168.0.165 in 0.025s
WARNING prana @ 192.168.0.165: Connection error occurred: prana @ 192.168.0.165: Connection lost
```

voed, 2024-10-21:
It does not matter, it shows only static memory and there is a lot of dynamic allocations that may lead to OOM

vpritulin, 2024-10-21:
I managed to get the verbose logs to work and as you can see, the humidity is incorrect "-90"

```
[17:16:32][C][:010]: Prana BLE Sensors
[17:16:32][C][prana_ble:068]: Prana BLE Switch 'Bluetooth connect'
[17:16:32][C][prana_ble:091]:   Restore Mode: always ON
[17:16:33][D][prana_ble:175]: [In/Out fans] update()
[17:16:33][D][prana_ble:118]: [Enable] update()
[17:16:33][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:33][D][prana_ble:118]: [Winter mode] update()
[17:16:33][D][prana_ble:175]: [Outlet fan] update()
[17:16:33][D][prana_ble:118]: [Heating] update()
[17:16:33][D][prana_ble:118]: [Fan lock] update()
[17:16:33][D][prana_ble:175]: [Inlet fan] update()
[17:16:34][D][prana_ble:175]: [In/Out fans] update()
[17:16:34][D][prana_ble:118]: [Enable] update()
[17:16:34][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:34][D][prana_ble:118]: [Winter mode] update()
[17:16:34][D][prana_ble:175]: [Outlet fan] update()
[17:16:34][D][prana_ble:118]: [Heating] update()
[17:16:34][D][prana_ble:118]: [Fan lock] update()
[17:16:34][D][prana_ble:175]: [Inlet fan] update()
[17:16:35][D][prana_ble:175]: [In/Out fans] update()
[17:16:35][D][prana_ble:118]: [Enable] update()
[17:16:35][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:35][D][prana_ble:118]: [Winter mode] update()
[17:16:35][D][prana_ble:175]: [Outlet fan] update()
[17:16:35][D][prana_ble:118]: [Heating] update()
[17:16:35][D][prana_ble:118]: [Fan lock] update()
[17:16:35][D][prana_ble:175]: [Inlet fan] update()
[17:16:36][D][prana_ble:175]: [In/Out fans] update()
[17:16:36][D][prana_ble:118]: [Enable] update()
[17:16:36][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:36][D][prana_ble:118]: [Winter mode] update()
[17:16:36][D][prana_ble:175]: [Outlet fan] update()
[17:16:36][D][prana_ble:118]: [Heating] update()
[17:16:36][D][prana_ble:118]: [Fan lock] update()
[17:16:36][D][prana_ble:175]: [Inlet fan] update()
[17:16:36][D][prana_ble:564]: [00:00:00:00:00:00] Not connected, will not send status.
[17:16:37][D][prana_ble:175]: [In/Out fans] update()
[17:16:37][D][prana_ble:118]: [Enable] update()
[17:16:37][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:37][D][prana_ble:118]: [Winter mode] update()
[17:16:37][D][prana_ble:175]: [Outlet fan] update()
[17:16:37][D][prana_ble:118]: [Heating] update()
[17:16:37][D][prana_ble:118]: [Fan lock] update()
[17:16:37][D][prana_ble:175]: [Inlet fan] update()
[17:16:38][D][prana_ble:175]: [In/Out fans] update()
[17:16:38][D][prana_ble:118]: [Enable] update()
[17:16:38][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:38][D][prana_ble:118]: [Winter mode] update()
[17:16:38][D][prana_ble:175]: [Outlet fan] update()
[17:16:38][D][prana_ble:118]: [Heating] update()
[17:16:38][D][prana_ble:118]: [Fan lock] update()
[17:16:38][D][prana_ble:175]: [Inlet fan] update()
[17:16:39][D][prana_ble:175]: [In/Out fans] update()
[17:16:39][D][prana_ble:118]: [Enable] update()
[17:16:39][D][prana_ble:118]: [Bluetooth connect] update()
[17:16:39][D][prana_ble:118]: [Winter mode] update()
[17:16:39][D][prana_ble:175]: [Outlet fan] update()
[17:16:39][D][prana_ble:118]: [Heating] update()
[17:16:39][D][prana_ble:118]: [Fan lock] update()
[17:16:39][D][prana_ble:175]: [Inlet fan] update()
[17:16:39][D][esp32_ble_client:110]: [0] [00:00:00:00:00:00] Found device
[17:16:39][D][esp32_ble_tracker:671]: Found device 00:00:00:00:00:00 RSSI=-72
[17:16:39][D][esp32_ble_tracker:692]:   Address Type: PUBLIC
[17:16:39][D][esp32_ble_tracker:694]:   Name: 'PRNAQaqPrana eco 200    '
[17:16:39][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[17:16:39][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[17:16:39][I][esp32_ble_client:067]: [0] [00:00:00:00:00:00] 0x00 Attempting BLE connection
[... 36 lines cut ...]
[17:16:42][D][esp32_ble_client:110]: [0] [00:00:00:00:00:00] ESP_GATTC_NOTIFY_EVT
[17:16:42][D][prana_ble:452]: [00:00:00:00:00:00] Notifying 10 children of latest status @0x3fff32ec.
[17:16:42][D][sensor:094]: 'prana Humidity': Sending state -28.00000 % with 0 decimals of accuracy
[17:16:42][D][sensor:094]: 'prana Pressure': Sending state 512.00000 hPa with 0 decimals of accuracy
[17:16:42][D][sensor:094]: 'prana TVOC': Sending state 16228.00000 ppb with 0 decimals of accuracy
[17:16:42][D][sensor:094]: 'prana CO2': Sending state 1022.00000 ppm with 0 decimals of accuracy
[17:16:42][D][switch:055]: 'Bluetooth connect': Sending state ON
[17:16:42][D][switch:055]: 'Enable': Sending state OFF
[17:16:42][D][switch:055]: 'Heating': Sending state OFF
[17:16:42][D][switch:055]: 'Winter mode': Sending state ON
[17:16:42][D][switch:055]: 'Fan lock': Sending state OFF
[17:16:42][D][fan:120]: 'In/Out fans' - Sending state:
[17:16:42][D][fan:121]:   State: OFF
[17:16:42][D][fan:123]:   Speed: 18
[17:16:42][D][fan:132]:   Preset Mode: \xec"
[17:16:42][D][fan:120]: 'Inlet fan' - Sending state:
[17:16:42][D][fan:121]:   State: OFF
[17:16:42][D][fan:123]:   Speed: 0
[17:16:42][D][fan:132]:   Preset Mode: \xec"
WARNING prana @ 192.168.0.165: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for prana @ 192.168.0.165
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for prana @ 192.168.0.165: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.0.165', port=6053))]: [Errno 111] Connect call failed ('192.168.0.165', 6053) (SocketAPIError)
INFO Trying to connect to prana @ 192.168.0.165 in the background
```

voed, 2024-10-21:
That's because the memory is corrupted. Show your config file.

voed, 2024-10-21:
Please try to change framework to esp-idf. I had the same problems on arduino framework, that's why i recommended to use IDF

vpritulin, 2024-10-21:
It works, but there is some interruption in the signal.

voed, 2024-10-21:
Looks like it is still crashes. Try to remove some components, leave only switches and fans. If that fixes a problem try to add them back one by one

vpritulin, 2024-10-21:
if I leave 1 sensor, then everything works, I start adding another 1, then the problem appears. And if you use humidity and tvoc separately. And the same if you add 2 sensors co2 and pressure, they work separately.

voed, 2024-10-21:
I see. I will try to reproduce it by myself, for now you can use `0.1.0-beta` branch, it should be stable for you

voed, 2024-10-21:
You can also try to change log level to warning, maybe it will solve the issue for now, but not sure.

voed, 2024-10-21:
Tried with your data that you provided earlier and it just works fine:

[image]

Try with other esp32 if you have one, clean build files and flash it by USB. If the issue still exists, try to get a fresh data packet from logs and i will test it.

vpritulin, 2024-10-23:
@voed Hello! I suggest closing this issue. It's most likely something with my ESP. After the last firmware it refused to work normally, but after it was disconnected for some time and when I reconnected it, everything works fine.

vpritulin, 2024-10-23:
in prana_ble: added update_interval: 30s to not overload esp too much

voed, 2024-10-23:
Got it. Thanks for feedback.

closed 2024-10-23 by voed

michal00111, 2024-10-23:
After reinstalling ESP32, everything works properly on 4 ESP32 and 4 recuperators. I'll check Auto mode later.