voed/esphome_prana_ble #15: Multiple devices - documentation update?

ondrejvysek, 2024-10-27:
I've followed the sample yaml config file, unfortunately I'm not able to get more devices (3) to work.

Line 5,6 all MAC addresses defined
Line 55 all MAC addresses defined
Line 62 all prana_ble clients defined

now I don't know how to extend sensor, switch, fan and number entries. e.g. for number tried:
```
number:
  - platform: prana_ble
    prana_ble_id: prana_client1
    name: Display brightness
  - platform: prana_ble
    prana_ble_id: prana_client2
    name: Display brightness   
```

but it seems this is not the right way. Is it possible to extend sample yaml to support multiple devices?

voed, 2024-10-27:
Sorry, forgot about that. Try number_fix branch (`github://voed/prana_ble@number_fix` in the component source).

ondrejvysek, 2024-10-27:
Thanks, no worries :) I can't see any difference, unfortunately. In the meantime, I was playing with the config, as I wanted to have a device name somewhere as it is under a single device in HA. here is what I have now, it seems work for me:
```
  mac_address_01: A8:B3:2A:79:C2:4E #Loznice
  mac_address_02: A8:BE:18:5F:12:C6 #Hoste
  mac_address_03: A0:B9:22:E8:0D:4A #Office
  mac_name_01: "Bedroom"
  mac_name_02: "Guest"
  mac_name_03: "Office"
```
sample sensor part (the same logic is used in switch, fan, number

```
sensor:
  - platform: prana_ble
    prana_ble_id: prana_client
    #temperature sensors
    temperature_inside_inlet: # top left sensor in Prana app
      name: "°C inside in ${mac_name_01}" 
    temperature_inside_outlet: # bottom left sensor in Prana app
      name: "°C inside out ${mac_name_01}" 
    temperature_outside_inlet:  # top right sensor in Prana app
      name: "°C outside in ${mac_name_01}"
    #other sensors
    co2:
      name: "CO2 ${mac_name_01}"
    pressure:
      name: "Pressure ${mac_name_01}"
    humidity:
      name: "Humidity ${mac_name_01}"
    tvoc:
      name: "TVOC ${mac_name_01}"
  - platform: prana_ble
    prana_ble_id: prana_client2
    #temperature sensors
    temperature_inside_inlet: # top left sensor in Prana app
      name: "°C inside in ${mac_name_02}" 
    temperature_inside_outlet: # bottom left sensor in Prana app
      name: "°C inside out ${mac_name_02}" 
    temperature_outside_inlet:  # top right sensor in Prana app
      name: "°C outside in ${mac_name_02}"
    #other sensors
    co2:
      name: "CO2 ${mac_name_02}"
    pressure:
      name: "Pressure ${mac_name_02}"
    humidity:
      name: "Humidity ${mac_name_02}"
    tvoc:
      name: "TVOC ${mac_name_02}"
  - platform: prana_ble
    prana_ble_id: prana_client3
    #temperature sensors
    temperature_inside_inlet: # top left sensor in Prana app
      name: "°C inside in ${mac_name_03}" 
    temperature_inside_outlet: # bottom left sensor in Prana app
      name: "°C inside out ${mac_name_03}" 
    temperature_outside_inlet:  # top right sensor in Prana app
      name: "°C outside in ${mac_name_03}"
    #other sensors
    co2:
      name: "CO2 ${mac_name_03}"
    pressure:
      name: "Pressure ${mac_name_03}"
    humidity:
      name: "Humidity ${mac_name_03}"
    tvoc:
      name: "TVOC ${mac_name_03}"
```

not sure if this is the right approach :)

voed, 2024-10-27:
I mean your config was okay, the problem was in the code.
```
number:
  - platform: prana_ble
    prana_ble_id: prana_client
    name: Display brightness

  - platform: prana_ble
    prana_ble_id: prana_client2
    name: Display brightness2
```
Like this it should work on that branch, at least it's compiled.

ondrejvysek, 2024-10-27:
Thanks, 
the longer sample is working for me even with the Master branch. I prefer having a name in the sensor rather than a number. 

I'll check how it is with the number_fox branch later

closed 2024-10-27 by ondrejvysek