voed/esphome_prana_ble #17: support for display content switching

mario-tux, 2024-10-31:
My recent Prana units can switch on the display content (fan speeds, temperature, clock,...) by using the bluetooth remote and the WiFi connection. What do you need to support this kind of functionality? Can I help?

voed, 2024-10-31:
Hi. I can add buttons to switch the display content, but i am not sure if there are any way to determine which content it currently showing, so you can select exactly what you want instead of just cycling between them. 
To find that out, change log level to `VERBOSE`

Add this buttons to your config so you will be able to switch between displays
```
button:
    - platform: template
      name: prana display left
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x19] 

    - platform: template
      name: prana display right
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x1A] 
```
Then provide packet log(see https://github.com/voed/esphome_prana_ble/issues/1#issuecomment-2271360261 for details) for every display and i will check if there are any useful data.

mario-tux, 2024-11-02:
I configured the to display buttons and they work well. Then I switched on the verbose logging to track the content of the received packets. I' hope they allow to create a sensor to feedback the display content. A derived control to select one of the available option (without multiple push of right/left button) would be fantastic. 

The log is available at: https://pastebin.com/S2xbCj2d
During this session I started with a "Date" content (the first in the app) and I cycled among all the options:
  Date -> Time  -> Fan -> Temperature inside -> Temperature outside -> CO2 -> VOC -> Humidity -> Pressure -> Date

voed, 2024-11-02:
Use `display` branch:

```
external_components:
  - source: "https://github.com/voed/esphome_prana_ble@display"
    components: [prana_ble] 
    refresh: 0s
```
Add select to your config:
```
select:
  - platform: prana_ble
    prana_ble_id: prana_client
    name: "Display mode"
```

Then, you will see the select entity in the HA
[image]
Use left/right buttons like in previous post and if select result changes, describe which number corresponds to the display, like 0 - date, 1 - time, etc

mario-tux, 2024-11-03:
I've tested the new `display` branch. These are the labels to associate to the numeric select:
- 9 - Date
- 10 - Time
- 0 - Fan
- 1 - Temperature indoor
- 2 - Temperature outdoor
- 3 - CO2
- 4 - VOC
- 5 - Humidity
- 6 - Air quality (?)
- 7 - Pressure

The label number 6 is a bit strange: I can't select it from the mobile app and, if selected using the esphome component, it show a "good" word on the display. It could be the air quality but also the filter state. I would hide it in the list if we are uncertain on the meaning.

voed, 2024-11-03:
Thanks!
> It could be the air quality but also the filter state.

I don't think there are any way to determine the filter state, so it should be air quality state based on both CO2 and VOC sensors. But i will add a remark in the config.

As i understand, in the mobile app there are no any option to make it cycle automatically(like every 5s), right? If so and if you are interested in such feature, you can create another PR and i will implement that.

mario-tux, 2024-11-03:
No, in the app there is no option for a periodic cycle. I don't think it would be useful as the unit makes a beep on each mode switch... :)

voed, 2024-11-04:
Check now, you should be able to se correct display mode and switch between them by select

voed, 2024-11-05:
Good. I will merge it to the main branch

closed 2024-11-05 by voed

mario-tux, 2024-11-08:
I harvested few other information about the strange display mode now labelled as "air quality". During live switching on the display it appears an "EFF" (maybe "efficiency"?) message followed by "GOOD" or "HIGH". On the app it corresponds to the "filtering" section. I would suggest to expose the mode naming it as "Filtering efficiency". 

I guess it is reported among the binary values in the packages: we could try to spot the position and report it as a sensor. Can I help in this?

voed, 2024-11-08:
Sure, create a new issue and try to catch logs with GOOD/HIGH values(and maybe other possible?) and i will se what i can do

PR #19 (2024-11-05): Select for switching between recuperator displays
Select what you want to display on your recuperator: "Fan", "Temp inside", "Temp outside", "CO2", "VOC", "Humidity", "Air quality(probably)", "Pressure", "Date", "Time"

Closes #17