voed/esphome_prana_ble #18: Crash using two units

mario-tux, 2024-11-03:
I know that @voed suggests to use one unit per ESP32 but it would be useful to be able to add at least two units. Maybe it is a bug in the code and not a limit of the memory. It there a way to test it?
I was able to use two units in the "daikin madoka ble" component (https://github.com/esphome/esphome/pull/5951)  jointly with one unit in the "prana ble" component in the same ESP32. On the other way I failed trying to use two units in the "plana ble" alone (removing the daikin ble component)..

This is the console log trying to use two units in "prana ble" component: https://pastebin.com/JxZWVYBm

voed, 2024-11-03:
As far as i have only one recuperator, it is really hard to troubleshoot that kind of problem. All that i can suggest for now is try to remove some components from your config (like sensors, display brightness etc) and see if that fixes the problem. If you find out what exactly crashes i can try to learn that deeply. But if it crashes randomly there's not much i can do.

brazoayeye, 2024-11-06:
Hello people, I don't have time to deepen the problem but I collected some infos, if they can be useful. 
I use this yaml 
[pranaterra.yaml.txt](https://github.com/user-attachments/files/17653404/pranaterra.yaml.txt)
and I have many crashes with its stack. I installed esphome in my laptop having the same problem, but now with stack decoded.

```
[22:44:51]I (28) boot: ESP-IDF 4.4.8 2nd stage bootloader
[22:44:51]I (29) boot: compile time 22:41:35
[22:44:51]I (29) boot: Multicore bootloader
[22:44:51]I (32) boot: chip revision: v1.0
[22:44:51]I (36) boot.esp32: SPI Speed      : 40MHz
[22:44:51]I (40) boot.esp32: SPI Mode       : DIO
[22:44:51]I (45) boot.esp32: SPI Flash Size : 4MB
[22:44:51]I (49) boot: Enabling RNG early entropy source...
[22:44:51]I (55) boot: Partition Table:
[22:44:51]I (58) boot: ## Label            Usage          Type ST Offset   Length
[22:44:51]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[22:44:51]I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[22:44:51]I (80) boot:  2 app0             OTA app          00 10 00010000 001c0000
[22:44:51]I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[22:44:51]I (95) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[22:44:51]I (103) boot: End of partition table
[22:44:51]I (107) boot: No factory image, trying OTA 0
[22:44:51]I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=319a0h (203168) map
[22:44:51]I (194) esp_image: segment 1: paddr=000419c8 vaddr=3ffbdb60 size=04f58h ( 20312) load
[22:44:51]I (202) esp_image: segment 2: paddr=00046928 vaddr=40080000 size=096f0h ( 38640) load
[22:44:51]I (219) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=e3988h (932232) map
[22:44:51]I (556) esp_image: segment 4: paddr=001339b0 vaddr=400896f0 size=121b4h ( 74164) load
[22:44:51]I (600) boot: Loaded app from partition at offset 0x10000
[22:44:51]I (635) boot: Set actual ota_seq=1 in otadata[0]
[22:44:51]I (635) boot: Disabling RNG early entropy source...
[22:44:51]I (646) cpu_start: Multicore app
[22:44:51]I (646) cpu_start: Pro cpu up.
[22:44:51]I (646) cpu_start: Starting app cpu, entry point is 0x400821d4
[22:44:51]I (0) cpu_start: App cpu up.
[22:44:51]I (666) cpu_start: Pro cpu start user code
[22:44:51]I (666) cpu_start: cpu freq: 160000000
[22:44:51]I (666) cpu_start: Application information:
[22:44:51]I (671) cpu_start: Project name:     pranaterra
[22:44:51]I (676) cpu_start: App version:      2024.10.2
[22:44:51]I (681) cpu_start: Compile time:     Nov  6 2024 22:38:42
[22:44:51]I (687) cpu_start: ELF file SHA256:  f55f41194ab2b6d4...
[22:44:51]I (693) cpu_start: ESP-IDF:          4.4.8
[22:44:51]I (698) cpu_start: Min chip rev:     v0.0
[22:44:51]I (703) cpu_start: Max chip rev:     v3.99
[22:44:51]I (707) cpu_start: Chip rev:         v1.0
[22:44:51]I (712) heap_init: Initializing. RAM available for dynamic allocation:
[22:44:51]I (719) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[22:44:51]I (725) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[22:44:51]I (731) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[22:44:51]I (738) heap_init: At 3FFCAE68 len 00015198 (84 KiB): DRAM
[22:44:51]I (744) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[22:44:51]I (750) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[22:44:51]I (757) heap_init: At 4009B8A4 len 0000475C (17 KiB): IRAM
[22:44:51]I (764) spi_flash: detected chip: generic
[22:44:51]I (768) spi_flash: flash io: dio
[22:44:51]I (774) cpu_start: Starting scheduler on PRO CPU.
[22:44:51]I (0) cpu_start: Starting scheduler on APP CPU.
[22:44:52][I][logger:156]: Log initialized
[22:44:52][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[22:44:52][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:44:52][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:44:52][I][app:029]: Running through setup()...
[22:44:52][C][esp32_ble:032]: Setting up BLE...
[22:44:52][C][wifi:048]: Setting up WiFi...
[22:44:52][D][esp-idf:000]: I (249) wifi:
[22:44:52][D][esp-idf:000]: wifi driver task: 3ffd61fc, prio:23, stack:6656, core=0
[22:44:52][D][esp-idf:000]:
[22:44:52]
[22:44:52][D][esp-idf:000][wifi]: I (1285) system_api: Base MAC address is not set
[... 161 lines cut ...]
WARNING Decoded 0x4008f9e9: esp_system_abort at C:\Users\giovanni\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:137
WARNING Decoded 0x40095a52: abort at C:\Users\giovanni\.platformio\packages\framework-espidf\components\newlib/abort.c:46
WARNING Decoded 0x40196dba: __wrap__Unwind_Resume_or_Rethrow
WARNING Decoded 0x401966af: __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90
WARNING Decoded 0x40196307: std::__throw_length_error(char const*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:78 (discriminator 2)
WARNING Decoded 0x40195bb6: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/basic_string.tcc:138
[... cut ...]
WARNING Decoded 0x401ad705: esphome::Component::call() at C:\Users\giovanni\Desktop\espHome\.esphome\build\pranaterra/src/esphome/core/component.cpp:104
WARNING Decoded 0x400e9a72: esphome::Application::setup() at C:\Users\giovanni\Desktop\espHome\.esphome\build\pranaterra/src/esphome/core/application.cpp:52 (discriminator 2)
WARNING Decoded 0x400ecca3: setup() at C:\Users\giovanni\Desktop\espHome\.esphome\build\pranaterra/src/main.cpp:753
WARNING Decoded 0x400d8c0f: esphome::loop_task(void*) at C:\Users\giovanni\Desktop\espHome\.esphome\build\pranaterra/src/esphome/components/esp32/core.cpp:67
```

Tell me if I have to test something else.

brazoayeye, 2024-11-07:
I don't know if it's related to multiple device or only a simple decoding error, in any case i changed that rows to 

  ```
if(this->fan_mode != data->fan_mode || this->preset_mode.empty())
  {
	if (data->fan_mode >= PRANA_FAN_MODES.size()) {
		ESP_LOGE(TAG, "Wrong PRANA_FAN_MODES %u", data->fan_mode);
	} else {
		this->preset_mode = PRANA_FAN_MODES[data->fan_mode];
		this->fan_mode = data->fan_mode;
		did_change = true;
	}
  }
```

Having 

```
[08:33:10][D][switch:055]: 'Winter mode Sala': Sending state OFF
[08:33:10][D][switch:055]: 'Fan lock Sala': Sending state OFF
[08:33:10][E][prana_ble:087]: Wrong PRANA_FAN_MODES 100
[08:33:10][D][fan:120]: 'Ventole Sala' - Sending state:
[08:33:10][D][fan:121]:   State: OFF
[08:33:10][E][prana_ble:087]: Wrong PRANA_FAN_MODES 100
[08:33:10][D][fan:120]: 'VentIN Sala' - Sending state:
[08:33:10][D][fan:121]:   State: OFF
[08:33:10][E][prana_ble:087]: Wrong PRANA_FAN_MODES 100
[08:33:10][D][fan:120]: 'VentOUT Sala' - Sending state:
[08:33:10][D][fan:121]:   State: OFF
```

Can you edit PRANA_FAN_MODES and PRANA_DISPLAY_MODES to avoid crashes during decoding problems? If you want to insert other debug infos I can test them to help you to undestand the problem.

voed, 2024-11-07:
Can you test now? I have added some checks and logging

brazoayeye, 2024-11-07:
compiling problem: missing , in LOGE

brazoayeye, 2024-11-07:
[pranaLog.txt](https://github.com/user-attachments/files/17662116/pranaLog.txt)

I copied locally the code and fixed the problem, i got the attached log.

```
[14:22:36][D][prana_ble:086]: New fan mode: 172
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 172
[14:22:36][D][prana_ble:086]: New fan mode: 172
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 172
[14:22:36][D][prana_ble:086]: New fan mode: 172
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 172
[14:22:36][D][number:012]: 'Luminosita Sala': Sending state 4.000000
[14:22:36][E][prana_ble:038]: Incorrect display mode: 255
[14:22:36][D][esp32_ble_client:110]: [0] [34:86:5D:0F:FF:16] ESP_GATTC_WRITE_DESCR_EVT
[14:22:36][D][esp32_ble_client:110]: [1] [24:D7:EB:6C:04:0A] ESP_GATTC_NOTIFY_EVT
[14:22:36][D][prana_ble:487]: [24:D7:EB:6C:04:0A] Notifying 10 children of latest status @0x3ffeac54.
[14:22:36][D][sensor:094]: 'Press Cucina': Sending state 512.00000 mmHg with 0 decimals of accuracy
[14:22:36][D][switch:055]: 'Bluetooth connect Cucina': Sending state ON
[14:22:36][D][switch:055]: 'Enable Cucina': Sending state OFF
[14:22:36][D][switch:055]: 'Heating Cucina': Sending state OFF
[14:22:36][D][switch:055]: 'Winter mode Cucina': Sending state OFF
[14:22:36][D][switch:055]: 'Fan lock Cucina': Sending state OFF
[14:22:36][D][prana_ble:086]: New fan mode: 100
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 100
[14:22:36][D][prana_ble:086]: New fan mode: 100
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 100
[14:22:36][D][prana_ble:086]: New fan mode: 100
[14:22:36][E][prana_ble:089]: Incorrect fan mode: 100
```

voed, 2024-11-07:
Hmm, it's weird that you are receiving this values. Can you change log level to verbose and provide the logs?

brazoayeye, 2024-11-07:
[pranaLog.txt](https://github.com/user-attachments/files/17664191/pranaLog.txt)

voed, 2024-11-07:
I see. Check master branch now, should be fine

brazoayeye, 2024-11-07:
For me it's solved. Let's check if the problem was related with [mario-tux](https://github.com/mario-tux) 's one

mario-tux, 2024-11-08:
Hi guys,
my last fixes I can run 3 units on the same ESP32. Great! 
I got few minor issues on the reported values and missing features but I will open other specific reports.

Thank you!

closed 2024-11-08 by mario-tux