voed/esphome_prana_ble #20: wrong unit_of_measurement for pressure: mmHg

brazoayeye, 2024-11-06:
in my Hass I recive the pressure in mmHg but it's showing hPa.
The solution is simple, add in the example:

```
    pressure:
      name: "Pressure"
      unit_of_measurement: mmHg
#      unit_of_measurement: hPa
#      filters:
#      - multiply: 1.3332
```

voed, 2024-11-06:
There's no reason to do that, you can chose whatever units you want directly in HA and it will be automatically converted
[image]

brazoayeye, 2024-11-06:
If you don't specify 
`unit_of_measurement: mmHg`
it used the standard UM hPa resulting in 750hPa ~ 560mmHg that's wrong. 

I agree that you can skip the filter part, that's commented, because you can do the same thing in HASS.

voed, 2024-11-06:
Oh, i see.  Should be fixed in the code now, just recompile your project.
Thanks.

closed 2024-11-06 by brazoayeye