voed/esphome_prana_ble #22: strange "TVOC" value

mario-tux, 2024-11-08:
One of my three units reports (on the app and on the display) I quiet high value for the TVOC sensor. I could be a real condition or a sensor requiring calibration: the component problem is that it reports a wrong (lower) value. For example: the TVOC value on unit/app is about 31000 but the component reports about 14600. I guess it is a problem in integer size or bytes involved in the conversion. If the problem is not clear (at your eyes) I can catch some binary packages from the logs.

voed, 2024-11-08:
I will check it, but yes, try to grab some logs.

mario-tux, 2024-11-08:
This is another unit with a correctly reported value of TVOC (same value in app and in component) of 2890:
```
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Temperature inside outlet': Received new state 19.200001
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Temperature inside outlet': Sending state 19.20000 °C with 1 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Temperature inside inlet': Received new state 16.500000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Temperature inside inlet': Sending state 16.50000 °C with 1 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Temperature outside outlet': Received new state 16.600000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Temperature outside outlet': Sending state 16.60000 °C with 1 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Temperature outside inlet': Received new state 15.700000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Temperature outside inlet': Sending state 15.70000 °C with 1 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Humidity': Received new state 79.000000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Humidity': Sending state 79.00000 % with 0 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - Pressure': Received new state 730.000000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - Pressure': Sending state 730.00000 mmHg with 0 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - TVOC': Received new state 2890.000000
[20:55:42][D][sensor:094]: 'VMC Master Bedroom - TVOC': Sending state 2890.00000 ppb with 0 decimals of accuracy
[20:55:42][V][sensor:043]: 'VMC Master Bedroom - CO2': Received new state 1261.000000
[20:55:43][D][sensor:094]: 'VMC Master Bedroom - CO2': Sending state 1261.00000 ppm with 0 decimals of accuracy
[20:55:43][D][switch:055]: 'VMC Master Bedroom - Fan lock': Sending state ON
[20:55:43][D][number:012]: 'VMC Master Bedroom - Display brightness': Sending state 4.000000
[20:55:43][D][select:015]: 'VMC Master Bedroom - Display mode': Sending state Fan (index 0)
[20:55:43][V][prana_ble:495]: Packet: BE.EF.05.01.67.2E.88.C4.5A.C0.01.C0.08.C0.00.C0.01.C0.00.00.00.C0.01.C0.01.C0.0A.40.01.40.0A.40.01.40.0A.00.00.80.00.00.00.C0.00.00.01.00.00.00.80.A5.00.80.9D.00.80.C0.00.80.A6.00.CF.84.ED.8B.4A.00.96.00.00.00.00.00.00.00.00.00.00.82.DA.00.00.00.00.00.00.00.80.02.00.00.00.00.00.FF.C0.00.00.00.00.00.13.00.8E.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
```

This is the unit with a high TVOC value (about 15900 at the moment on the app) with a value in component show as 914:
```
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Temperature inside outlet': Received new state 19.700001
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Temperature inside outlet': Sending state 19.70000 °C with 1 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Temperature inside inlet': Received new state 17.700001
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Temperature inside inlet': Sending state 17.70000 °C with 1 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Temperature outside outlet': Received new state 18.000000
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Temperature outside outlet': Sending state 18.00000 °C with 1 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Temperature outside inlet': Received new state 16.200001
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Temperature outside inlet': Sending state 16.20000 °C with 1 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Humidity': Received new state 75.000000
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Humidity': Sending state 75.00000 % with 0 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - Pressure': Received new state 730.000000
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - Pressure': Sending state 730.00000 mmHg with 0 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - TVOC': Received new state 914.000000
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - TVOC': Sending state 914.00000 ppb with 0 decimals of accuracy
[20:55:50][V][sensor:043]: 'VMC Secondary Bedroom - CO2': Received new state 526.000000
[20:55:50][D][sensor:094]: 'VMC Secondary Bedroom - CO2': Sending state 526.00000 ppm with 0 decimals of accuracy
[20:55:50][D][number:012]: 'VMC Secondary Bedroom - Display brightness': Sending state 4.000000
[20:55:50][D][select:015]: 'VMC Secondary Bedroom - Display mode': Sending state VOC (index 4)
[20:55:50][V][prana_ble:495]: Packet: BE.EF.05.01.67.2E.7A.D2.5A.C0.01.C0.08.C0.00.C0.00.C0.00.00.00.C0.01.C0.01.C0.14.40.01.40.14.40.01.40.14.00.00.80.00.00.00.C0.00.00.01.00.00.00.80.B1.00.80.A2.00.80.C5.00.80.B4.00.CB.82.0E.C3.92.00.AA.00.00.00.00.00.00.00.00.00.00.82.DA.00.00.00.00.00.00.00.80.01.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.F3.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
```

voed, 2024-11-09:
Very strange. Try to grab more logs because now i don't see any sense in this numbers, like data packet is corrupted or recuperator just reports wrong values. Maybe you can test if CO2 sensor have the same behaviour?

mario-tux, 2024-11-09:
The CO2 readings are always the same on the unit and on the component. It looks like there is a wrap around at about 15000. The CO2 readings are under this threshold.

mario-tux, 2024-11-09:
What are the bytes involved in the TVOC value?

voed, 2024-11-09:
Its 63-64 bytes(C3 92 from your last log), then you need to swap them and apply 0x3fff mask like [here](https://github.com/voed/esphome_prana_ble/blob/master/esphome/components/prana_ble/sensor/prana_ble_sensors.cpp#L69)
I have no idea why they did it, this protocol is kinda stupid and there's no even checksum to make sure the packet is correct.

mario-tux, 2024-11-09:
> Very strange. Try to grab more logs because now i don't see any sense in this numbers, like data packet is corrupted or recuperator just reports wrong values. Maybe you can test if CO2 sensor have the same behaviour?

The mask used (0x3fff) implies that the TVOC and CO2 value can't be greater than 16383. I managed to capture another log making one unit to read high TVOC values (using gas from an alcool bottle...). It is difficult to catch from the display the value corresponding to the package on the log. 
During the transmission of the following log the display shows "TVOC = (about) 27000".

```
[14:10:54][V][sensor:043]: 'VMC Office - Temperature inside outlet': Received new state 20.700001
[14:10:54][D][sensor:094]: 'VMC Office - Temperature inside outlet': Sending state 20.70000 °C with 1 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - Temperature inside inlet': Received new state 20.000000
[14:10:54][D][sensor:094]: 'VMC Office - Temperature inside inlet': Sending state 20.00000 °C with 1 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - Temperature outside outlet': Received new state 20.400000
[14:10:54][D][sensor:094]: 'VMC Office - Temperature outside outlet': Sending state 20.40000 °C with 1 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - Temperature outside inlet': Received new state 19.700001
[14:10:54][D][sensor:094]: 'VMC Office - Temperature outside inlet': Sending state 19.70000 °C with 1 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - Humidity': Received new state 46.000000
[14:10:54][D][sensor:094]: 'VMC Office - Humidity': Sending state 46.00000 % with 0 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - Pressure': Received new state 728.000000
[14:10:54][D][sensor:094]: 'VMC Office - Pressure': Sending state 728.00000 mmHg with 0 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - TVOC': Received new state 10848.000000
[14:10:54][D][sensor:094]: 'VMC Office - TVOC': Sending state 10848.00000 ppb with 0 decimals of accuracy
[14:10:54][V][sensor:043]: 'VMC Office - CO2': Received new state 8178.000000
[14:10:54][D][sensor:094]: 'VMC Office - CO2': Sending state 8178.00000 ppm with 0 decimals of accuracy
[14:10:54][D][number:012]: 'VMC Office - Display brightness': Sending state 4.000000
[14:10:54][D][select:015]: 'VMC Office - Display mode': Sending state VOC (index 4)
[14:10:54][V][prana_ble:495]: Packet: BE.EF.05.01.67.2F.6D.6A.5A.C0.01.C0.08.C0.00.C0.00.C0.00.00.00.C0.01.C0.01.C0.14.40.01.40.14.40.01.40.14.00.00.80.00.00.00.C0.00.00.01.00.00.00.80.C8.00.80.C5.00.80.CF.00.80.CC.00.AE.DF.F2.EA.60.00.00.00.00.00.00.00.00.00.00.00.00.82.D8.00.00.00.00.00.00.00.80.00.00.00.00.00.00.FF.C0.00.00.00.00.04.13.00.2E.E9.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (137)
```

The CO2 value can be computed from bytes `DF.F2` as `0xDFF2 & 0x3FFF = 0x1FF2 = 8178` (the swap is not necessary reading if we look at the data packet representation). The TVOC value should correspond to the following two bytes `EA.60` so the actual computation is `0xEA60 & 0x3FFF = 0x2A60 = 10848`. My guess is that we should use (at least for the TVOC value) a different mask: `0xEA60 & 0x7FFF = 0x6A60 = 27232`. This extended mask doesn't look to work for the CO2: in this example would bring a wrong value. 

My proposal is to extend the mask only for the TVOC value. Let me know if it makes sense for you.

voed, 2024-11-09:
Looks like you are right. Tested with some data from previous reports and it seems fine. Can you make a PR?

mario-tux, 2024-11-09:
ok, I'll provide a PR.

closed 2024-11-09 by voed