voed/esphome_prana_ble #23: missing "night" and "boost" mode

mario-tux, 2024-11-08:
In my units I got two extra modes: "night" and "boost" mode. It looks in an exclusive mode with manual, auto and auto+ modes so we could add them to such list in the fan modes. Do you have enough information from the reverse engineering of the bluetooth protocol to support command launch and gathering feedback information or you you need extra packages capture?

voed, 2024-11-08:
This modes just changes fan speed to min and max, so i don't see much reason to implement that. But if you find them useful i can add them of course.

mario-tux, 2024-11-08:
I understand your point but it would be just to clone the app functionalities in the component. One is then free to enable and use such controls. 

My 2 cents.

voed, 2024-11-08:
At least on my device it's just a buttons to change fan speed, so i can impelent it as buttons.

mario-tux, 2024-11-08:
The logic is similar to the fan modes: manual, auto, auto+, night or boost.

voed, 2024-11-09:
I mean there is no any way to track that "boost" or "night" state, so if i make that it will conflict: when in auto mode and fan speed is 1 it will show night mode instead.

mario-tux, 2024-11-09:
Ok. I think you can ignore this request. It is not very useful.

closed 2024-11-09 by mario-tux