voed/esphome_prana_ble #25: [suggestion] 2 fan controls instead of 3

mario-tux, 2024-11-08:
I was wondering if we can keep the component functionalities and usability if we reduce the exposed fan controls from (In/Out fans -- Inlet fan -- Outlet fan) to just (Inlet fan -- Outlet fan). If the "Fan lock" toggle is active the two fans should be "in sync" and changing one (mode or speed) should be reflected on the other; it the toggle is disabled they should independently work (like now).

Maybe the actual mode mimics the protocol and it is easier to manage for you. I was just wondering if it could be a better and minimal way to expose the component features. Feel free to close this report if you don't like the idea.

voed, 2024-11-08:
I think it may be a little bit confusing because there's no way to make other fan unavailable when fans locked. But i don't like the current implementation as well, so i will think about it.

mario-tux, 2024-11-08:
At the moment you can't change the fans speed in the two inlet and outlet fans if the "fan lock" toggle is active. Is this a consequence of the protocol or you did it?

voed, 2024-11-08:
A bit of both i guess. Did not tested what will happen in that case, so just added some checks to not let that happen. How that works in the Prana app?

mario-tux, 2024-11-08:
I guess the behavior in the app is the same for all the models. If the "fan lock" is active on one fan speed control is shown. If it is disabled two independent speed controls are provided. 
My "two fans" suggested layout looks minimal enough: if the lock is active you should force the changes on both the fan controls. If disabled the two fans should be managed in an independent way (like now).