voed/esphome_prana_ble #27: Fan lock switch

matlar83, 2024-11-11:
When I toogle the "fan lock" switch,

it toggle by itself other 2 times:
https://github.com/user-attachments/assets/d140a818-d99e-4c97-b3ff-a5c4ad81a112

The final result is correct, but - because I use the switch value to show or hide a UI tab - it produces a flickering UI.

voed, 2024-11-11:
Can you check now? It's still may happen sometimes, but very rarely.

matlar83, 2024-11-11:
I tried the new version, it is for sure an improvement.
It still happen, even if not every time. But it is not so rare to happen, I would say 20% of the time.

voed, 2024-11-11:
Try to reduce update interval to 0.4-0.5s:
```
prana_ble:
  - id: prana_client
    ble_client_id: prana
    update_interval: 0.5s

```
Should be better. But cannot get rid of this completely right now, it needs a deep rework to fix that.

matlar83, 2024-11-11:
Seems better, it's fully usable. Thank you

mario-tux, 2024-11-22:
A too fast polling rate is not good for wireless channel pollution (2.4GHz): I would look to a better solution keeping the pooling rate low (every 5 o 10 seconds).

voed, 2024-11-22:
It's fine. The packet size is only 130 bytes, so even polling 10 times per second will generate only ~100kb/s of traffic. I don't think its a problem.