voed/esphome_prana_ble #28: Fan pecenteage level

matlar83, 2024-11-11:
On My Prana VMC, the fan levels are 0 to 5.

As I try to set the fan value on HA, each level is represented by a 10% percentage:
level 0 --> 0%
level 1 --> 10%
level 2 --> 20%
level 3 --> 30%
level 4 --> 40%
level 5 --> 50%

Any value above 50%, will go to a 50% percentage:
https://github.com/user-attachments/assets/6fbf8ac8-488f-49b2-938f-f6039d81a91c

voed, 2024-11-11:
And below 50% it is controlled correctly?

matlar83, 2024-11-11:
Yes, it is controlled correctly below 50%.
It works, but on a scale from 0 to 50%.

voed, 2024-11-11:
Try with this branch:
```
external_components:
  - source: "https://github.com/voed/esphome_prana_ble@fan_speed_count"
    components: [prana_ble] 
    refresh: 0s
```
You will be able to change speed count for each fan like this:

```
  - platform: prana_ble
    prana_ble_id: prana_client
    has_auto: true
    name: Inlet fan
    speed_count: 6
    fan_type: INLET
```
But it might be confusing because 100%/6 will give floating percent values in HA
And if level 0 still a valid speed and not disabled fan for you, set speed count to 7

matlar83, 2024-11-11:
Yes, it works perfectly :-)
Thank you!

voed, 2024-11-11:
Merged to the main branch

closed 2024-11-11 by voed