voed/esphome_prana_ble #31: Wifi instead of bluetooth

Gruffuss, 2024-11-21:
Do you think you can make this work via wifi instead of bluetooth at some point?

brazoayeye, 2024-11-21:
I read on https://github.com/zauan/hassio-prana (i think) that the wifi connection is encrypted and can't be reversed engeneered. I asked many times to info@prana24.com to develop an integration for esp home or, at least, to open a local web server (or socket server) to handle local requests.

Maybe more people harassing prana service could give us more possibilities

Gruffuss, 2024-11-21:
Then i should start doing that i  just got 2 of them installed today, both are WiFi enabled.

voed, 2024-11-21:
Why don't just use bluetooth? Or they are wifi only?

Gruffuss, 2024-11-21:
No, they are not WiFi on and i am going to use bluetooth for now, but frankly... I sadly don't have the available sockets to plug esp32s close to the prana devices

mario-tux, 2024-11-22:
> No, they are not WiFi only and i am going to use bluetooth for now, but frankly... I sadly don't have the available sockets to plug esp32s close to the prana devices

I got 3 units with bluetooth+wifi and will disable wifi when all the local setup using the esphome+HA will work fine. I don't see the point in keeping enabled it. I will never update the firmware after such switch: it could just break the bluetooth compatibility.

The plug for the esp32 can stay even at 6-10 meters away from the Prana unit. Don't worry!

brazoayeye, 2024-11-22:
Wifi range is longer than ble one. If they could open something over wifi it's meaningless to develope it using espHome

closed 2025-05-15 by voed