voed/esphome_prana_ble #32: [bounty] Inclusion in Esphome official project

mario-tux, 2024-11-22:
Hi, I appreciate the valuable work of @voed on his external esphome component. As already discussed on the Telegram channel, it would be great if his work would be included in the upstream project. This would guarantee in the future a great continuity without any risk of breakage across major releases of the main project. This requires further work: finalization of the features and of design of the exposed controls, code cleanup, few documentation and creation of a PR. The acceptance is far to be guaranteed and it could take months...

Given this, I would personally  appreciate such kind of effort and I'm willing the repay it. I already did a small donation but I will add a further donation (+40$) on the PR acceptance.

If other people is interested could using this report to declare other similar willing.

voed, 2024-11-22:
Hi. Thanks for your offer, but i don't think its possible in near future, and that's why:
1. I am not a C++ programmer. Yes, i can write some code that works, but the code quality may vary due to lack of experience. Also there's lots of code standards that i need to learn to make that possible.
2. There's a lot of device revisions with different sensors, fans, displays etc. Even if the communication protocol is the same(well, compatible at least), i need to own all of that devices to maintain the component properly.
3. The current implementation was developed for ~4 years due to my laziness and lack of time.
4. I hate python and even more i hate Esphome's config validator. There is no any documentation for it, only examples from other components, so i think config validation stuff took more time than actual C++ code of the component. And there's no any feedback either, i asked a question in the Esphome's discussions and didn't got any answer, that's why fan's config still looks ugly and complicated.

mario-tux, 2024-11-22:
I understand your points: if it is fine for you, we can keep this post as publicity. Maybe someone is interested in helping. The donation/bounty is open to others.

Furthermore this post can collect other potential donation promises in other to promote the contribution on this goal.