voed/esphome_prana_ble #33: Control lost

Burtila, 2025-02-06:
[logs_bedroom-bluetooth_run (1).txt](https://github.com/user-attachments/files/18694178/logs_bedroom-bluetooth_run.1.txt)

Hi, 
I don`t understand why the connection is lost some times or in other times i set a value for the fan, but the action is not executed.
The ESP is at max 2 meter distance.

Some times when i change fast the speed,  it works as i expected.

Let me know if other information is needed.

voed, 2025-02-06:
Hi. From this log i cannot say anything, it may be poor connection between esp32 and your wifi router or just some problem with a code that forces ESP to reboot. Can you connect your ESP via usb cable and get the uart logs?

voed, 2025-02-07:
Looks like it's wifi connection issues. What board do you use?

Burtila, 2025-02-07:
The log is from WSP32-wroom-32U, i had the same issue with WSP32-wroom-32

[image]

voed, 2025-02-07:
That's weird. Can you try to relocate the board somewhere around the room? Maybe something is creating a noise in current area. 
Also you can try to increase update interval:
```
prana_ble:
  - id: prana_client
    ble_client_id: prana
    update_interval: 10s
```
Just for testing at least.
Also maybe try to create an access point on your PC/smarphone and connect esp through it to make sure its not a router settings or something

Burtila, 2025-02-07:
1. as i have check the logs via wifi, I don't consider to be a wifi issue because i got response from esp all the time, only the Bluetooth connection looks with issue.
2. i have add "a new" Bluetooth ESP (same device new configuration), this time without adding the sensor that i have mentioned in post 3, the CO2 sensor and temperature and now it works. It`s passing all my test that  fail previously.
3. Taking point 1 and 2 in consideration if the solution for point 1 is not found, i suggest to write on the first page, that is suggested to use the ESP only for PRANA control and not to add other sensor to the ESP based on this issue.
4. Thank you for  your involvement and for this integration!

voed, 2025-02-07:
> i have add "a new" Bluetooth ESP (same device new configuration), this time without adding the sensor that i have mentioned in post 3, the CO2 sensor and temperature and now it works. It`s passing all my test that fail previously.

Oh, i thought you mean Prana's CO2 and temperature sensor. If it's some custom sensor devices yes it might cause some issues depends on their code and polling interval. Because according to the logs it was  like you are sending too many messages to the home assistant. Theoretically it should work, so if you post your config i will try to figure out what's wrong with it.

closed 2025-03-11 by voed