voed/esphome_prana_ble #37: using mqtt the fan speeds mapping has changed (?)

mario-tux, 2025-04-25:
I was used to connect the ESP32 with HA with MQTT (for a series of reasons). It worked well until few days ago. Now the speed mapping become wrong: 10%->10% ; 20%->20% ; 30%->30% ; 40%->80% (!!!!) ; 50%->80% (!!!!) ; 60%->80% (!!!!) ; 70%->70% ; 80%->80% ; 90%->90% ; 100%->100% 

As you can see the 40-60% choices are wrongly mapped. Using the HA api this is not problem. Can you reproduce the problem?

mario-tux, 2025-04-25:
switching from api to mqtt and backward... I'm facing the problem using the api too... I need more testing. Any idea?

mario-tux, 2025-04-25:
This is my config:
```
esphome:
  name: "esp-prana-bridge"
  friendly_name: "esp-prana-bridge"

esp32:
  #board: esp32-poe-iso
  board: wt32-eth01
  framework:
    type: esp-idf
    #version: 5.1.2 #Optionally, you can even specify a newer IDF version. It may contain some fixes and work better in some cases.
    #platform_version: 6.5.0
    #sdkconfig_options:
    #  CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

#esp32_ble_tracker:
#  scan_parameters:
#    active: true
  
#bluetooth_proxy:
#  active: true

logger:
  #level: VERBOSE

api:
  encryption:
    key: !secret hassio_api_aes_key

#mqtt:
#  broker: !secret hassio_mqtt_hostname
#  username: !secret hassio_mqtt_username
#  password: !secret hassio_mqtt_password

ota:
  - platform: esphome
    password: !secret hassio_api_pwd

#wifi:
#  ssid: !secret hassio_wifi_ssid
#  password: !secret hassio_wifi_psk

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

button:
  - platform: restart
    name: "Restart"
    
external_components:
  - source: github://voed/prana_ble@master
    components: [ prana_ble ] 
    refresh: 0s

esp32_ble_tracker:

esp32_ble:
  io_capability: display_yes_no

ble_client:
  - mac_address: *****
    id: prana_vmc_study
  - mac_address: ******
    id: prana_vmc_master_bedroom
[... 169 lines cut ...]
#                 id: prana_vmc_study
#                 service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
#                 characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
#                 value: [0xbe, 0xef, 0x04, 0x1A] 

select:
  - platform: prana_ble
    prana_ble_id: prana_vmc_client_study
    display_mode:
      name: "VMC Study - Display mode"
    fan_mode:
      name: "VMC Study - Fan mode"
  - platform: prana_ble
    prana_ble_id: prana_vmc_client_master_bedroom
    display_mode:
      name: "VMC Master Bedroom - Display mode"
    fan_mode:
      name: "VMC Master Bedroom - Fan mode"
  - platform: prana_ble
    prana_ble_id: prana_vmc_client_secondary_bedroom
    display_mode:
      name: "VMC Secondary Bedroom - Display mode"
    fan_mode:
      name: "VMC Secondary Bedroom - Fan mode"
```

mario-tux, 2025-04-25:
This is a log from the ESP32: the fans was at 20% and I selected the 40% for the Inlet/Outlet fans (using an action):
```
[21:14:03][D][fan:021]: 'VMC Study - In/Out fans' - Setting:
[21:14:03][D][fan:024]:   State: ON
[21:14:03][D][fan:030]:   Speed: 4
[21:14:03][W][prana_ble:046]: Setting fan speed 4
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][fan:120]: 'VMC Study - In/Out fans' - Sending state:
[21:14:03][D][fan:121]:   State: ON
[21:14:03][D][fan:123]:   Speed: 4
[21:14:03][W][component:239]: Component api took a long time for an operation (52 ms).
[21:14:03][W][component:240]: Components should block for at most 30 ms.
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][fan:021]: 'VMC Study - In/Out fans' - Setting:
[21:14:03][D][fan:024]:   State: ON
[21:14:03][D][fan:030]:   Speed: 7
[21:14:03][W][prana_ble:046]: Setting fan speed 7
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][fan:120]: 'VMC Study - In/Out fans' - Sending state:
[21:14:03][D][fan:121]:   State: ON
[21:14:03][D][fan:123]:   Speed: 7
[21:14:03][W][component:239]: Component api took a long time for an operation (75 ms).
[21:14:03][W][component:240]: Components should block for at most 30 ms.
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][fan:021]: 'VMC Study - In/Out fans' - Setting:
[21:14:03][D][fan:024]:   State: ON
[21:14:03][D][fan:030]:   Speed: 10
[21:14:03][W][prana_ble:046]: Setting fan speed 10
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][prana_ble:322]: BEEF040C
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_UP success
[21:14:03][D][fan:120]: 'VMC Study - In/Out fans' - Sending state:
[21:14:03][D][fan:121]:   State: ON
[21:14:03][D][fan:123]:   Speed: 10
[21:14:03][W][component:239]: Component api took a long time for an operation (75 ms).
[21:14:03][W][component:240]: Components should block for at most 30 ms.
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][fan:021]: 'VMC Study - In/Out fans' - Setting:
[21:14:03][D][fan:024]:   State: ON
[21:14:03][D][fan:030]:   Speed: 8
[21:14:03][W][prana_ble:046]: Setting fan speed 8
[21:14:03][D][prana_ble:322]: BEEF040B
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_DOWN success
[21:14:03][D][prana_ble:322]: BEEF040B
[21:14:03][D][prana_ble:300]: [7C:9E:BD:B4:17:62] writing button CMD_FAN_SPEED_DOWN success
[21:14:03][D][fan:120]: 'VMC Study - In/Out fans' - Sending state:
[21:14:03][D][fan:121]:   State: ON
[21:14:03][D][fan:123]:   Speed: 8
[21:14:03][W][component:239]: Component api took a long time for an operation (77 ms).
[21:14:03][W][component:240]: Components should block for at most 30 ms.
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:03][D][esp32_ble_client:191]: [0] [7C:9E:BD:B4:17:62] ESP_GATTC_WRITE_CHAR_EVT
[21:14:05][D][esp32_ble_tracker:282]: Starting scan...
[21:14:08][D][prana_ble:322]: BEEF0501000000005A
[21:14:08][D][esp32_ble_client:191]: [2] [34:AB:95:FB:7F:0A] ESP_GATTC_NOTIFY_EVT
[21:14:08][D][prana_ble:487]: [34:AB:95:FB:7F:0A] Notifying 12 children of latest status @0x3ffe8048.
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Temperature inside outlet': Sending state 20.80000 °C with 1 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Temperature inside inlet': Sending state 18.80000 °C with 1 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Temperature outside outlet': Sending state 17.20000 °C with 1 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Temperature outside inlet': Sending state 15.50000 °C with 1 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Humidity': Sending state 72.00000 % with 0 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - Pressure': Sending state 721.00000 mmHg with 0 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - TVOC': Sending state 25730.00000 ppb with 0 decimals of accuracy
[21:14:08][D][sensor:093]: 'VMC Secondary Bedroom - CO2': Sending state 569.00000 ppm with 0 decimals of accuracy
[... cut ...] 00000
[21:14:08][D][select:015]: 'VMC Secondary Bedroom - Display mode': Sending state Fan (index 0)
[21:14:08][D][select:015]: 'VMC Secondary Bedroom - Fan mode': Sending state Manual (index 0)
[21:14:09][D][esp32_ble_client:191]: [2] [34:AB:95:FB:7F:0A] ESP_GATTC_WRITE_CHAR_EVT
[21:14:09][W][component:239]: Component esp32_ble took a long time for an operation (110 ms).
[21:14:09][W][component:240]: Components should block for at most 30 ms.
[21:14:10][D][prana_ble:322]: BEEF0501000000005A
```

mario-tux, 2025-04-26:
I was wondering what could have caused such a change: my unit got a wifi interface and it is connected to the cloud. Sometimes I use the mobile app but very rarely. In the past, the app notified me about the availability of new firmware and asked me to upgrade the unit (by the app) in order to access the unit (in other words, it forced me in order to continue to use the app). In the past month I didn't manually upgrade any firmware on the Prana units. I'm wondering if they changed the upgrade policy, allowing the unit to upgrade by itself without any manual configuration. This would explain the behavior changing. Actually, the reported firmware is 3.2.16 (I don't remember the past versions).The change could be in the packet reported by the unit; such a change could be present in the new units so it could be worthy of support. In order to track the problem, I simplified my ESPHome firmware, removing 2 of the 3 units and I switched on the debug output on the logs.

In the first log (https://pastebin.com/YjiFz77W), I rebooted the ESP32 and the prana unit (study) was active at speed 3. Then I used the app to change the speed to 5. As you can see, the ESP32 has autonomously started to change the speed with higher rates until a landing value of 8.

In the second log (https://pastebin.com/gMPqHwn5), I rebooted the ESP32 and the prana unit was active on speed 1 and I used HA to switch on speed 4. Again, the final speed is 8. I'm wondering if this could be related to the recent commit reworking the fan logic. I always use the "Fan Inlet/Outlet" keeping active the fan lock. Switching off the fan lock, it looks like I'm able to separately control the fans even on the "problematic" speed (4-7).

voed, 2025-04-26:
Hi. I see that state is sent multiple times:

```
[11:13:06][D][fan:120]: 'VMC Study - Inlet fan' - Sending state:
[11:13:06][D][fan:121]:   State: ON
[11:13:06][D][fan:123]:   Speed: 5
[11:13:06][D][fan:120]: 'VMC Study - Outlet fan' - Sending state:
[11:13:06][D][fan:121]:   State: ON
[11:13:06][D][fan:123]:   Speed: 5
[11:13:06][D][fan:120]: 'VMC Study - In/Out fans' - Sending state:
[11:13:06][D][fan:121]:   State: ON
[11:13:06][D][fan:123]:   Speed: 5
```
I gues this is the reason why it behaves like this. I have some ideas why it happening, but it may be very painful to fix that and needs like global rework of the component, which i can't do for now due to lack of time. But alternatively, we can try "new" fan control protocol which is should be available for your device. It was discussed [here](https://github.com/voed/esphome_prana_ble/issues/5#issuecomment-2282806577): commands `beef0420` and up should set fan speed directly. Can you test that? See [this post](https://github.com/voed/esphome_prana_ble/issues/5#issuecomment-2277739096) for more information about that or feel free to ask me here or in telegram PM if you have any questions. 
This method of fan control should be much more reliable and it's easy to implement.

mario-tux, 2025-04-26:
> Hi. I see that state is sent multiple times:
>
> ```
> [11:13:06][D][fan:120]: 'VMC Study - Inlet fan' - Sending state:
> [11:13:06][D][fan:121]:   State: ON
> [11:13:06][D][fan:123]:   Speed: 5 [...]

I noted the involving for the 3 HA fans (inlet, outlen and inlen/outlet) in some logs but in the latter a sent using pastebin, they are not involved. Is it inline with your idea of the problem?

Could you argue about it? I don't understand because I didn't have problem until few days ago and, without updating the ESP32 units or HA, it broke...

> But alternatively, we can try "new" fan control protocol which is should be available for your device. It was discussed [...]

I'm not sure what do you want me to try? the direct raw commands using some kind of app? a git branch of your project?

voed, 2025-04-26:
> I noted the involving for the 3 HA fans (inlet, outlen and inlen/outlet) in some logs but in the latter a sent using pastebin, they are not involved. Is it inline with your idea of the problem?

Yeah, and that is even more weird. I will try to test it further.

> I'm not sure what do you want me to try? the direct raw commands using some kind of app? a git branch of your project?

Yes, get "ble scanner" app on your smartphone, connect to the recuperator and try to send this commands:
```
beef0420
beef0421
beef0422
beef0423
beef0424
beef0425
beef0426
beef0427
beef0428
beef0429
beef042a
beef042b
beef042c
beef042d
beef042e
beef042f
beef0430
beef0431
...
etc
```
Some of them should change the speed of inlet/outlet and both fans

Alternatively, you can add buttons to your config:
```
button:
    - platform: template
      name: prana 0x20
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x20] 

button:
    - platform: template
      name: prana 0x21
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x21] 
...
etc
```
But i guess app method is easier

mario-tux, 2025-04-26:
Ok, it was easy:
```
beef041f -> 1-* (set only inlet)
beef0420 -> 2-*
beef0421 -> 3-*
beef0422 -> 4-*
beef0423 -> 5-*
beef0424 -> 6-*
beef0425 -> 7-*
beef0426 -> 8-*
beef0427 -> 9-*
beef0428 -> 10-*
beef0429 -> *-1  (set only outlet)
beef042a -> *-2
beef042b -> *-3
beef042c -> *-4
beef042d -> *-5
beef042e -> *-6
beef042f -> *-7
beef0430 -> *-8
beef0431 -> *-9
beef0432 -> *-10
beef0433 -> 1/10 (set both with lock)
beef0434 -> 2/10
beef0435 -> 3/10
beef0436 -> 4/10
beef0437 -> 5/10
beef0438 -> 6/10
beef0439 -> 7/10
beef043a -> 8/10
beef043b -> 9/10
beef043c -> 10/10
```

voed, 2025-04-26:
Nice, thanks! I will try to create a branch for tests in few hours.
Can you please test if there are any more commands after 0403C? Maybe we can also control brightness or display mode like that

mario-tux, 2025-04-26:
> Nice, thanks! I will try to create a branch for tests in few hours. 

Thanks!

> Can you please test if there are any more commands after 0403C? Maybe we can also control brightness or display mode like that

I continued until `beef0442` without any reaction from the prana unit: I guess there no more commands.

voed, 2025-04-26:
> I continued until `beef0442` without any reaction from the prana unit: I guess there no more commands.

I see, thanks.
Try `direct_fan` branch, add `direct_control: true` to the fan configuration and it should work:
```
fan:
  - platform: prana_ble
    prana_ble_id: prana_client
    direct_control: true
    fan_in:
      name: Inlet fan
    fan_out:
      name: Outlet fan
    fan_both:
      name: In/Out fans # use fan_lock switch to control separate fans
```

mario-tux, 2025-04-26:
I tested the new direct method to control the fan and it looks to work (as the previous). In other words, the problem with the speed rate among 40% and 70% remains: it is unrelated. 
This is my config (https://pastebin.com/6DycQF6q) if you want to check. The branch compiled regularly.

In this log (https://pastebin.com/FXdmN06M) I had the fans speed at 30% then I used the app (so using the cloud) to switch to 50%: it remained as this few seconds, then the ESP32 it looks decided to change the speed (why?!). It went up to 100% and then back to 80%. I'm getting crazy: if I use wifi to change the speed, why the ESP32 does change it? The same if I use HA to change the speed.

voed, 2025-04-26:
Looks super strange for me.  Are you sure you don't have any automations enabled or something? Check the logbook in home assistant to make sure the control was triggered by esphome. And i will try to test what happens when the device is controlled from other sources like wifi or remote control, maybe that's the problem

mario-tux, 2025-04-27:
I did more testing on my environment: I'm sure at 70% that the issue come from matter-hub service (https://github.com/t0bst4r/home-assistant-matter-hub) I use to locally expose (some) devices to my Google Nest Hub Max. I can select with a label witch devices to expose and the Prana units are NOT exposed but, in some way, it could touch something unexpected on the unexposed devices. It is an Alpha state software...

I'm saying it could be the issue because it upgraded 3 days ago and during my tests today I restarted its docker container and the issue stopped! Now I can use the "crazy fan speeds" without problem.

Sorry for the noise. At least you tested the direct fan speed control: do you think are going to commit it as an optional feature? Is it supported by all the units? 

You can close his issues.

closed 2025-04-27 by voed