voed/esphome_prana_ble #41: Buttons and sliders do not work properly

catdogmaus, 2025-04-29:
Buttons and sliders do not work properly, they always revert back to default setting. 
Even more confusion adds that sometimes they do switch device to requeued setting but actually do show default in HA shortly after so you never know what it actually is.

Sensors show correct information.

I do operate device that should actually be encrypted but it still connects to ESPHome whatever pin for device is set up.

voed, 2025-04-30:
Hi. First of all please make some screenshots from the Prana app, maybe your device somehow differs from the others.
Also you need to get a debug log from the esp32(ESPHome Builder - your device - LOGS button). Especially when you are trying to change the speed.
If you have some other components in your ESP32 like bluetooth proxy or some display or speaker - remove them, there may be lack of resources or just some conflict
And share your config.

catdogmaus, 2025-04-30:
I use Prana RC app, that was standard when I got the device.
I have literally every other option avail in HA but no "speed"!
I do know almost nothing about Bluetooth protocols but I am sure that my device uses (at least partially) different UUIDs then in your code. 
When look at logs it is seen that device reports its state but almost no commands gets thru. My hypothesis, its those UUIDs and that pesky encryption thing. I did test it bit and there seems to be no difference how ESP works right now whatever pin I set up in app.

[image]
[image]
[image]

[logs_esphome-web-3310ac_logs.txt](https://github.com/user-attachments/files/19981623/logs_esphome-web-3310ac_logs.txt)

[esphome-web-3310ac.txt](https://github.com/user-attachments/files/19981644/esphome-web-3310ac.txt)

voed, 2025-05-01:
As long as you received any info from recuperator, the protocol should be fine.
[19:30:41][D][fan:021]: 'In/Out fans' - Setting:
[19:30:41][D][fan:024]:   State: OFF
[19:30:41][W][prana_ble:028]: Cannot control this fan. Check fan lock switch
In the log you are trying to control both fans while your fan_lock switch is turned off. You should change state of the fan lock switch or control inlet and outlet fans separately, like in the prana app. Also note that enabling auto mode also prevents manual fan control.
If it's still does not work, try to change direct_control to false. It should be false by default, i will update example config.

catdogmaus, 2025-05-01:
> try to change direct_control to false.

It was false, supposedly for older devices so I did change it for testing.

Fan buttons work in manual mode, (except In/Out fans) challenge is to change manual or auto+. Sometimes it is really unresponsive and not registering change. In logs there is often failed commands
`[19:27:25][D][prana_ble:357]: BEEF0501000000005A
[19:27:26][D][prana_ble:357]: BEEF0501000000005A
[19:27:26][D][esp-idf:000][BTU_TASK]: E (28453635) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.`
It was really hard to notice what is working bcs logs are changing almost every second. interval. 

**What is certainly not working** properly is bluetooth button. It is sort of working - I notice that connection seems to get lost but button always reverts immediately back to ON regardless what actual state is. 
Not working is display mode - mode changes to something but mostly not that it should be, after that it always reverts back in HA desktop to fan mode. I suppose there must be some mismatch between actual available modes and in code ones. I don't know what is "air quality", my device has "Efficiency". 
When you device has different modes it would be maybe reasonable to set up modes in yaml. It creates more flexibility between different models.
I am also missing "night mode" and vent speeds but I don't know what exactly should be available.
thanks

voed, 2025-05-02:
> Fan buttons work in manual mode, (except In/Out fans) challenge is to change manual or auto.

I assume that your device has no auto+ mode. I forgot about this. I will fix that, you can check the progress in #43 

> Not working is display mode - mode changes to something but mostly not that it should be, after that it always reverts back in HA desktop to fan mode. I suppose there must be some mismatch between [...]

For this, you can create a new issue, but again it seems like your device just does not reports the display mode(always reports 0)

> I am also missing "night mode" and vent speeds but I don't know what exactly should be available.

It just sets fan speed to minimum and maximum, so i didn't implemented this because you can do the same by fan controls.

> When you device has different modes it would be maybe reasonable to set up modes in yaml. It creates more flexibility between different models.

Well that's very good idea, but i don't have enough statistic to create that configs. Also i think it's very difficult for user to understand which exactly model he owns. There's no any markings on the device itself and no any info in the bluetooth protocol, so the only way is to read on the recuperator box(which i threw away long time ago)

I am aware about lack of documentation, i just don't have time for now to create readable wiki with detailed explanation of all settings and requirements

catdogmaus, 2025-05-02:
All separate problems have their own issues opened so I can close this now

closed 2025-05-02 by catdogmaus