voed/esphome_prana_ble #42: Prana configuration gets lost from ESP device

catdogmaus, 2025-04-30:
Every time my esp32 devices power gets interrupted all the Prana configuration gets lost. Device boots up just as basic clean device and prana stuff needs to be written again.
I have never encountered something like that before. Have 8 other esp devices working perfectly.

Have tested with esp32 dev board V1 and esp32 C3 supermini with exactly same results.

voed, 2025-05-01:
That's very weird behaviour, looks like incorrect partition table of the esphome.
When it rolls back, are you flashing it again via OTA? If so, try to flash by cable. Use "Factory" binary format.
Also you can try to switch framework from esp-idf to arduino, maybe it will help somehow.

catdogmaus, 2025-05-01:
I had question in mind why just Prana code, and with different boards? Is the code somehow different?
Anyway, after 6 or 7 rewrites it seems to be OK, at least in moment.

closed 2025-05-01 by catdogmaus