voed/esphome_prana_ble #45: Not working is display mode

catdogmaus, 2025-05-02:
Not working display mode - mode changes to something but mostly not that it should be, after that it always reverts back in HA desktop to (default) fan mode. 
I suppose there must be some mismatch between actual available modes and in code ones. I don't know what is "air quality", my device has instead "Efficiency".

In logs I can see that it tries to do something but after dozens of unsuccessful attempts eventually queue gets full and it throws
`[21:18:23][D][esp-idf:000][BTU_TASK]: E (37315842) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.`

voed, 2025-05-02:
To troubleshoot that, you need to change log level to VERBOSE in your config, change the mode few time and send me the log. Also you can switch it few times from the remote or wifi, just for confidence.
To minimize the log flood you can remove other components like switches, sensors etc. All you need is a display select and prana_ble itself.

> I suppose there must be some mismatch between actual available modes and in code ones. I don't know what is "air quality", my device has instead "Efficiency".

That's okay, i was not sure about that mode name, it needs to be renamed. Thanks for report.

> In logs I can see that it tries to do something but after dozens of unsuccessful attempts eventually queue gets full and it throws
[21:18:23][D][esp-idf:000][BTU_TASK]: E (37315842) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

I guess it should be fine too, but you can reduce update interval:
```
prana_ble:
  - id: prana_client
    ble_client_id: prana
    update_interval: 5s
```
Of course, it will reduce the polling rate and responsiveness of some controls, so you can play a bit with this option to achieve the best experience for you.

catdogmaus, 2025-05-03:
[logs_esphome-web-logs.txt](https://github.com/user-attachments/files/20024453/logs_esphome-web-logs.txt)

voed, 2025-05-03:
I see lots of bluetooth disconnects in your log, maybe you should try to put your esp closer to the recuperator. But, as i said, your device does not provides any info about current display mode. You can add template buttons to set desired mode as described [here ](https://github.com/voed/esphome_prana_ble/issues/17#issuecomment-2450045216), like in the prana app or remote control.
Also maybe you can try to check for firmware updates with Prana Online 2.0 app if it works for you

catdogmaus, 2025-05-04:
I am not expert but it looks for me that it refuses to communicate with my device
[logs_esphome-web-logs.txt](https://github.com/user-attachments/files/20028529/logs_esphome-web-logs.txt)

voed, 2025-05-04:
In this log connection is completely stable, there's no disconnects or packet losses like in previous one. But i don't see any changes in the display mode

catdogmaus, 2025-05-05:
I somehow closed this

voed, 2025-05-05:
Does your device have IR remote? Try to change the display mode from it and record the log.

catdogmaus, 2025-05-05:
[logs_esphome-web-3310ac_logs (1).txt](https://github.com/user-attachments/files/20041357/logs_esphome-web-3310ac_logs.1.txt)

> But i don't see any changes in the display mode

I need to have clarification here - any changes is in logs(!?), bcs when I scroll right or left actual device screen scrolls along without problems.
Anyway this should be with remote - when anything shows up.

catdogmaus, 2025-05-06:
> there's nothing i can do with that.

Well, theoretically there is probably option to make it so that when device does not report state back just ignore it and assume that latest command was successful with displaying that state.
I do not know why with main software it did not display correctly but scroll left/right I saw change every time.

voed, 2025-05-06:
The problem is there's no exact command to set desired display mode. I am using left/right commands multiple times until it reaches desired mode.
But you still can add left/right buttons:
``` yaml
button:
    - platform: template
      name: prana display left
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x19] 

    - platform: template
      name: prana display right
      on_press:
        then:
            - ble_client.ble_write:
                id: prana # your ble_client id
                service_uuid: 0000baba-0000-1000-8000-00805f9b34fb
                characteristic_uuid: 0000cccc-0000-1000-8000-00805f9b34fb
                value: [0xbe, 0xef, 0x04, 0x1A] 
```
It's less convenient, but better than nothing i guess.

catdogmaus, 2025-05-07:
OK good to know

closed 2025-05-15 by voed