voed/esphome_prana_ble #49: New sub-device support (improvement)

mario-tux, 2025-07-16:
Hi,
looking at the new release of esphome (https://esphome.io/changelog/2025.7.0), I was wondering the new feature "sub-device" could be used to expose the `n` recuperator units that one can configure on a single ESP32 board as different "sub-devices" with the related features/controls. This would expose a more clear separations of them: now they are unified in a, quiet confusing, bunch.

matlar83, 2025-08-01:
Hi,
it does not seems to work properly:

```
select:
  - platform: prana_ble
    device_id: VMC_camera_matrimoniale
    prana_ble_id: prana_camera_matrimoniale
    # select what to display on your recuperator screen: "Fan", "Temp inside", "Temp outside", "CO2", "VOC", "Humidity", "Air quality", "Pressure", "Date", "Time"
    display_mode:
      name: "Display mode" 
    # switch between Auto, Auto+ and Manual modes
    fan_mode:
      name: "Fan mode"
```

I got the following error: 

> [device_id] is an invalid option for [select.prana_ble]. Please check the indentation.

matlar83, 2025-08-04:
Oh, thank you! It works like a charm

mario-tux, 2025-08-05:
Hi. 
I'm starting to use the new sub-device feature of esphome with your component. The necessity to use `device_id` for each entity is annoying but manageable. Now I'm stuck in another (even more) annoying drawback: the generation of the entity names. 
This is my configuration: https://pastebin.com/5Rr1nGbd

[image]

With this configuration the generate (unique) names and friendly names are not natural:

[image]

I would get name like: fan.vmc_master_bedroom_inlet_fan, fan.vmc_secondary_bedroom_inlet_fan, ....
The linter warns me if I try to name the entity like "Inlet_fan" for duplicate names:

[image]

closed 2026-10-22 by voed