voed/esphome_prana_ble #50: HA complaining about ppb

catdogmaus, 2025-07-26:
I noticed that HA is complaining about measurement unit. 
`(<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; expected one of ['mg/m³', 'µg/m³']; `
As far as I remember Prana is using ppb (Parts per billion) but HA has other ideas about that. So now I am bit confused. ppb is real unit so not sure why HA does not like it.
As far as I know 1ppb is 1 microgram per liter.

voed, 2025-07-27:
Changed device class to VOLATILE_ORGANIC_COMPOUNDS_PARTS, should be fine now. Thanks for report.

voed, 2025-07-28:
Oops. Should be fixed now.